moveit/moveit_tutorials #613: Compile error in moveit_cpp_tutorial:  ‘moveit_cpp’ was not declared in this scope; did you mean ‘moveit_cpp_ptr’?

robotrapta, 2021-03-17:
### Description

I'm getting a compiler error when following the [getting started](https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html) instructions for the tutorial on a clean Ubuntu 20.04+Noetic machine.  (Arm64 FWIW)

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04 Linux ip-172-31-3-122 5.4.0-1039-aws #41-Ubuntu SMP Wed Feb 24 23:14:54 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
* Source or Binary build?  Source
* If binary, which release version?
* If source, which git commit or tag?  186f9c464351c2105daece39bbddce586f381308

### Steps to reproduce

On a fresh EC2 graviton instance with Ubuntu 20.04, install Noetic, and follow the tutorial setup instructions https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html

When I get to `catkin build` it crashes.  Same results regardless of whether I include MoveIt source or not.

### Expected behaviour
Successful build

### Backtrace or Console output

```
Errors     << moveit_tutorials:make /home/ubuntu/ws_moveit/logs/moveit_tutorials/build.make.000.log                                           
/home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp: In function ‘int main(int, char**)’:
/home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp:34:42: error: ‘moveit_cpp’ was not declared in this scope; did you mean ‘moveit_cpp_ptr’?
   34 |   auto moveit_cpp_ptr = std::make_shared<moveit_cpp::MoveItCpp>(nh);
      |                                          ^~~~~~~~~~
      |                                          moveit_cpp_ptr
/home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp:34:30: error: parse error in template argument list
   34 |   auto moveit_cpp_ptr = std::make_shared<moveit_cpp::MoveItCpp>(nh);
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp:34:67: error: no matching function for call to ‘make_shared<<expression error> >(ros::NodeHandle&)’
   34 |   auto moveit_cpp_ptr = std::make_shared<moveit_cpp::MoveItCpp>(nh);
      |                                                                   ^
In file included from /usr/include/c++/9/memory:81,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:23,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /opt/ros/noetic/include/ros/forwards.h:37,
                 from /opt/ros/noetic/include/ros/common.h:37,
                 from /opt/ros/noetic/include/ros/ros.h:43,
                 from /home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp:1:
/usr/include/c++/9/bits/shared_ptr.h:714:5: note: candidate: ‘template<class _Tp, class ... _Args> std::shared_ptr<_Tp> std::make_shared(_Args&& ...)’
  714 |     make_shared(_Args&&... __args)
      |     ^~~~~~~~~~~
/usr/include/c++/9/bits/shared_ptr.h:714:5: note:   template argument deduction/substitution failed:
/home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp:34:67: error: template argument 1 is invalid
   34 |   auto moveit_cpp_ptr = std::make_shared<moveit_cpp::MoveItCpp>(nh);
      |                                                                   ^
/home/ubuntu/ws_moveit/src/moveit_tutorials/doc/moveit_cpp/src/moveit_cpp_tutorial.cpp:37:35: error: parse error in template argument list
   37 |   auto planning_components = std::make_shared<moveit_cpp::PlanningComponent>(PLANNING_GROUP, moveit_cpp_ptr);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [doc/moveit_cpp/CMakeFiles/moveit_cpp_tutorial.dir/build.make:63: doc/moveit_cpp/CMakeFiles/moveit_cpp_tutorial.dir/src/moveit_cpp_tutorial.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4801: doc/moveit_cpp/CMakeFiles/moveit_cpp_tutorial.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:141: all] Error 2
cd /home/ubuntu/ws_moveit/build/moveit_tutorials; catk
```

robotrapta, 2021-03-17:
Looks like 186f9c464351c2105daece39bbddce586f381308 might have caused this?  Rolling back to edb533c14e38b4f4ec34345263a390dc59bb3a6c makes the error go away.

@henningkayser

rhaschke, 2021-03-18:
Yes, #549 caused this, which goes in sync with https://github.com/ros-planning/moveit/pull/2127, https://github.com/ros-planning/moveit_msgs/pull/95, and https://github.com/ros-planning/moveit_resources/pull/47. Until the next release is out, you will need to build _all_ those packages from source to use the latest versions. Alternatively, rollback moveit_tutorials as suggested in https://github.com/ros-planning/moveit_tutorials/issues/613#issuecomment-801304192.

henningkayser, 2021-03-18:
@robotrapta sorry for the inconvenience

robotrapta, 2021-03-18:
No worries!  I understand that multi-package upgrades are tricky, but necessary for long-term code health.

I have what I need.  Maybe leave this open until the release is done so it's easier for others to find the workaround?  Do search engines prioritize open issues vs closed?  IDK.

GoBeyond2017, 2021-03-25:
if you look up the source code of the  MoveItCpp class, you will find the MoveItCpp's namespace is moveit::planning_interface not moveit_cpp, so replace the moveit_cpp with moveit::planning_interface, then it works.

lucbettaieb, 2021-04-21:
Leaving a comment to share that it ain't great that the solution for fixing *tutorial code* is to wait for it to sync up with other released packages that as of over a month later, still don't seem to be done.

If the namespaces aliased, I'm throwing in a +1 for reopening and merging https://github.com/ros-planning/moveit_tutorials/pull/614 so that future travelers aren't tripped up by this.  Then, when things are synced, just change the namespacing back.  

Or at minimum, leave a temporary note in the example code...