moveit/moveit_tutorials #615: Quick question about fake_execution flag in moveIt package

ranjitkathiriya, 2021-03-29:
Hello there,

Just a quick question, if we are configuring a real robot with move it what will be the flag of fake_execution flag? it will be true or false?

include file="$(find kuka_moveit_roof)/launch/move_group.launch">
      <arg name="publish_monitored_planning_scene" value="true" />
       <arg name="allow_trajectory_execution" value="true"/>
      <arg name="fake_execution" value="true"/>
    <arg name="info" value="true"/>
    <arg name="debug" value="$(arg debug)"/>
    <arg name="pipeline" value="$(arg pipeline)"/>
    <arg name="load_robot_description" value="$(arg load_robot_description)"/>
  </include>

and what args is important when we are configuring a real robot? In repository, I only find 'publish_monitored_planning_scene' can you please put a light on this question?

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04

closed 2021-03-29 by ranjitkathiriya

felixvd, 2021-03-29:
It is good practice to answer your own questions if you found the solution.

ranjitkathiriya, 2021-03-29:
Sorry! but I think that I have posted a question in a wrong place so I have moved my question to Moveit https://github.com/ros-planning/moveit/issues/2570, instead of moveit tutorials. 

once again sorry for inconvenience.