moveit/moveit_tutorials #619: IndexError: list assignment index out of range in Python Move group interface

99arp, 2021-04-05:
### Description
I think there is a small error in tutorial file for python move group interface in planning to a joint goal in this  [line](https://github.com/ros-planning/moveit_tutorials/blob/d2458c1ca66461240717a66eb7ff54fcdb318b84/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L187).  
Overview of your issue here.

### Your environment
* ROS Distro:[Melodic]

* OS Version: e.g. Ubuntu 20.04
* Source 

The array has a length of 6, so the Index error occurs while trying to access joint_goal[6] = 0 

### Backtrace or Console output
` joint_goal[6] = 0
IndexError: list assignment index out of range
`
Can fix this if required.

v4hn, 2021-04-06:
The `panda_arm` group has 7 joints (`panda_joint[1,2,3,4,5,6,7]`) and the array is 7 entries long over here.
I'm not sure why you would see a list with length 6 here.

99arp, 2021-04-06:
Sorry, I mixed up a couple of things. Closing this issue.

closed 2021-04-06 by 99arp