moveit/moveit_tutorials #623: E: Unable to locate package ros-noetic-opw-kinematics-plugin

mohamedsayed18, 2021-04-09:
### Description
E: Unable to locate package ros-noetic-opw-kinematics-plugin

### Your environment
* ROS Distro: noetic
* OS Version: e.g. Ubuntu 20.04.2 LTS

### Steps to reproduce
sudo apt install ros-noetic-opw-kinematics-plugin

### Expected behaviour
successful installation

### Backtrace or Console output

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-noetic-opw-kinematics-plugin
```

gavanderhoorn, 2021-04-09:
Looks like it hasn't been released: [status_page/ros_noetic_default.html?q=opw](http://repositories.ros.org/status_page/ros_noetic_default.html?q=opw).

Perhaps @JeroenDM would be willing to do that.

burakaksoy, 2022-05-16:
I have the same issue

mohamedsayed18, 2022-05-16:
Just clone clone this repo to your workspace https://github.com/Jmeyer1292/opw_kinematics

closed 2022-05-16 by mohamedsayed18

gavanderhoorn, 2022-05-16:
As you can see at [status_page/ros_noetic_default.html?q=opw](http://repositories.ros.org/status_page/ros_noetic_default.html?q=opw), everything now has been released, and should be available.

Provided you are using Melodic or Noetic, and a supported architecture of course.

In that case there should be no need to build anything from source any more.

simonschmeisser, 2022-05-16:
The name of the package is `ros-noetic-moveit-opw-kinematics-plugin` actually, I opened a PR

burakaksoy, 2022-05-16:
To build from source these commands worked:
```
sudo apt install ros-noetic-ros-industrial-cmake-boilerplate
sudo apt install ros-noetic-opw-kinematics
cd catkin_ws/src/
git clone --recursive https://github.com/JeroenDM/moveit_opw_kinematics_plugin.git
catkin_make
```
For the released packages just running 
```
sudo apt install ros-noetic-moveit-opw-kinematics-plugin
``` 
works.