moveit/moveit_tutorials #638: UR dependencies not declared in package.xml

v4hn, 2021-05-04:
The new dependencies on universal robot package for the `mesh_filter` tutorial [are commented out in the `package.xml`](https://github.com/ros-planning/moveit_tutorials/blob/master/package.xml#L63-L66). As explained in [a comment of the PR that created the new mesh_filter tutorial](https://github.com/ros-planning/moveit_tutorials/pull/558/files#diff-37a67ff78eb7260214b323353263b9af40a2fa98719a1e81937fae0159df87a3), officially declaring the UR dependencies as `run_depend` will make rosdep fail in CI:

```
rosdep install -y -q -n --from-paths . --ignore-src --rosdistro noetic
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
moveit_tutorials: Cannot locate rosdep definition for [ur_description]
The command "rosdep install -y -q -n --from-paths . --ignore-src --rosdistro noetic" failed [trial 1 of 3].
```

This is because the UR packages are not released for noetic [and not even melodic](https://github.com/ros-industrial/universal_robot/issues/448).
We could add the source repository of these packages to CI to resolve the error, but they are only runtime dependencies and not even used in tests.
Alternatively we could disable the broken keys in the CI check, but I'm not sure whether we should do that and how the old/new CI could be changed to do it.

rhaschke, 2021-05-04:
IMHO, our tutorials shouldn't depend on packages that are not released/available. 
Why, in the first place, was the mesh_filter tutorial based on the UR? Can't we ask the ros_industrial consortium for a release?

v4hn, 2021-05-04:
> IMHO, our tutorials shouldn't depend on packages that are not released/available. [...]
> Can't we ask the ros_industrial consortium for a release?

I tend to agree, but the universal_robotics repository is a special case and @gavanderhoorn can probably comment.

> Why, in the first place, was the mesh_filter tutorial based on the UR?

I'm ok with that (which is why I merged it), but nobody commented [on a similar statement](https://github.com/ros-planning/moveit_tutorials/pull/558#issuecomment-829515640) in the PR.