moveit/moveit_tutorials #640: openrave docker issue

UmutUz, 2021-05-05:
I read in earlier discussions that openrave should be working fine with Melodic over docker and thought why not try it out. Well it doesn't work for Ubuntu 18/ROS Melodic (at least not for me). This is where it stops: 

`rosrun moveit_kinematics auto_create_ikfast_moveit_plugin.sh --iktype Transform6D $MYROBOT_NAME.urdf <planning_group_name> <base_link> <eef_link>`

```
Step 3/3 : RUN pip install sympy==0.7.1
 ---> Running in 26c61a662d58
Downloading/unpacking sympy==0.7.1
  Cannot fetch index base URL https://pypi.python.org/simple/
  Could not find any downloads that satisfy the requirement sympy==0.7.1
Cleaning up...
No distributions at all found for sympy==0.7.1
Storing debug log for failure in /root/.pip/pip.log
The command '/bin/sh -c pip install sympy==0.7.1' returned a non-zero code: 1
```

Not sure if it's docker or me.

rhaschke, 2021-05-05:
> `Cannot fetch index base URL https://pypi.python.org/simple`

Do you have a working network connection?
sympy 0.7.1 is still available on pypi: https://pypi.org/project/sympy/0.7.1/

UmutUz, 2021-05-05:
If your question is docker related, I cannot answer that. I just followed the instructions and this was the error I got. General network connection (as in connecting to the internet), yes I do but I have no clue if I have to change some docker settings.

rhaschke, 2021-05-05:
If you have a network connection in your host, usually docker can connect to the network as well. But, please, test the network connection from within your docker container, e.g. pinging some known site like google.

UmutUz, 2021-05-05:
Sorry for the late answer, forgot a lot about docker after not using it for a while.

So I can ping google.com with no problem.

rhaschke, 2021-05-06:
Duplicate of https://github.com/ros-planning/moveit/issues/2649. Closing here.

closed 2021-05-06 by rhaschke