moveit/moveit_tutorials #642: How to create a tiago_moveit_config folder by own using MoveIT Setup Assistant?

hjw-1014, 2021-05-14:
### Description

Currently, I want to create a new **tiago_moveit_config** folder. I found a Tiago xacro file [here](https://github.com/pal-robotics/tiago_dual_robot/blob/kinetic-devel/tiago_dual_description/robots/tiago_dual.urdf.xacro), and I tried to load it using **MoveIT Setup Assistant**, but I got problems, and I don't find a solution after I searched.

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04

### Steps to reproduce
Basically, follow [this](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/setup_assistant/setup_assistant_tutorial.html).

1. Using `roslaunch moveit_setup_assistant setup_assistant.launc` to open MoveIT Setup Assistant.
2. Load Tiago xacro, then errors come.

```
Gtk-Message: 06:54:24.305: GtkDialog mapped without a transient parent. This is discouraged.
resource not found: tiago_description_calibration
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/huangjiawei/ws_moveit/src/geometric_shapes
ROS path [2]=/home/huangjiawei/ws_moveit/src/moveit/moveit
ROS path [3]=/home/huangjiawei/ws_moveit/src/moveit_msgs
ROS path [4]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/moveit_planners
ROS path [5]=/home/huangjiawei/ws_moveit/src/moveit/moveit_plugins/moveit_plugins
ROS path [6]=/home/huangjiawei/ws_moveit/src/moveit_resources/moveit_resources
ROS path [7]=/home/huangjiawei/ws_moveit/src/moveit_resources/fanuc_description
ROS path [8]=/home/huangjiawei/ws_moveit/src/moveit_resources/fanuc_moveit_config
ROS path [9]=/home/huangjiawei/ws_moveit/src/moveit/moveit_commander
ROS path [10]=/home/huangjiawei/ws_moveit/src/moveit_resources/panda_description
ROS path [11]=/home/huangjiawei/ws_moveit/src/moveit_resources/panda_moveit_config
ROS path [12]=/home/huangjiawei/ws_moveit/src/moveit_resources/pr2_description
ROS path [13]=/home/huangjiawei/ws_moveit/src/moveit/moveit_core
ROS path [14]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/chomp/chomp_motion_planner
ROS path [15]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/chomp/chomp_optimizer_adapter
ROS path [16]=/home/huangjiawei/ws_moveit/src/moveit_resources/prbt_ikfast_manipulator_plugin
ROS path [17]=/home/huangjiawei/ws_moveit/src/moveit_resources/prbt_moveit_config
ROS path [18]=/home/huangjiawei/ws_moveit/src/moveit_resources/prbt_pg70_support
ROS path [19]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/moveit_ros
ROS path [20]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/occupancy_map_monitor
ROS path [21]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/planning
ROS path [22]=/home/huangjiawei/ws_moveit/src/moveit/moveit_plugins/moveit_fake_controller_manager
ROS path [23]=/home/huangjiawei/ws_moveit/src/moveit/moveit_kinematics
ROS path [24]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/ompl
ROS path [25]=/home/huangjiawei/ws_moveit/src/moveit_resources/prbt_support
ROS path [26]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/move_group
ROS path [27]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/manipulation
ROS path [28]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/perception
ROS path [29]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/robot_interaction
ROS path [30]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/warehouse
ROS path [31]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/benchmarks
ROS path [32]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/planning_interface
ROS path [33]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/chomp/chomp_interface
ROS path [34]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/visualization
ROS path [35]=/home/huangjiawei/ws_moveit/src/moveit/moveit_runtime
ROS path [36]=/home/huangjiawei/ws_moveit/src/moveit/moveit_ros/moveit_servo
ROS path [37]=/home/huangjiawei/ws_moveit/src/moveit/moveit_setup_assistant
ROS path [38]=/home/huangjiawei/ws_moveit/src/moveit/moveit_plugins/moveit_simple_controller_manager
ROS path [39]=/home/huangjiawei/ws_moveit/src/moveit/moveit_plugins/moveit_ros_control_interface
ROS path [40]=/home/huangjiawei/ws_moveit/src/panda_moveit_config
ROS path [41]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/pilz_industrial_motion_planner_testutils
ROS path [42]=/home/huangjiawei/ws_moveit/src/moveit/moveit_planners/pilz_industrial_motion_planner
ROS path [43]=/home/huangjiawei/ws_moveit/src/rviz_visual_tools
ROS path [44]=/home/huangjiawei/ws_moveit/src/moveit_visual_tools
ROS path [45]=/home/huangjiawei/ws_moveit/src/moveit_tutorials
ROS path [46]=/opt/ros/melodic/share 
```

### Expected behaviour
Could anyone tell me if it's because I use the wrong tiago xacro file or how can Ioad a tiago in MoveIT Setup Assistant to create my own tiago_moveit_config folder?

Thanks in advance.

rhaschke, 2021-05-15:
As stated by the error message, your workspace is missing the package `tiago_description_calibration`. Instead of using the repo https://github.com/pal-robotics/tiago_dual_robot, I suggest starting with https://github.com/pal-robotics/tiago_robot (assuming you have a single robot only). 
Note, that there is a ready-to-use https://github.com/pal-robotics/tiago_moveit_config package available.
Finally, for questions regarding `tiago_description` you should consider https://answers.ros.org or the PAL sites.

closed 2021-05-15 by rhaschke

hjw-1014, 2021-05-17:
> As stated by the error message, your workspace is missing the package `tiago_description_calibration`. Instead of using the repo https://github.com/pal-robotics/tiago_dual_robot, I suggest starting [...]
Thank you so much, I'm new to MoveIt, so very sorry about this.