moveit/moveit_tutorials #654: Running TrajOpt tutorial

gautz, 2021-07-06:
### Description
Trying to run TrajOpt tutorial https://ros-planning.github.io/moveit_tutorials/doc/trajopt_planner/trajopt_planner_tutorial.html

### Your environment
* ROS Distro: [noetic]
* OS Version: Ubuntu 20.04
* Source build from Master a8c57d39

### Steps to reproduce
Setup a workspace and run the demo following https://ros-planning.github.io/moveit_tutorials/doc/trajopt_planner/trajopt_planner_tutorial.html
- moveit_planners_trajopt is not built because of ``moveit/moveit_planners/trajopt/CATKIN_IGNORE``
- I tried to checkout trajopt_ros and tesseract manually to the commits given here: https://github.com/ros-planning/moveit/pull/1593#issue-303175526 . No success: trajopt_examples fail to build with lots of pcl-1.10 errors

### Expected behaviour
- Either erase the tutorial as long as https://github.com/ros-planning/moveit/pull/1626#issuecomment-533918473 is not merged
- Or detail custom workspace setup steps in the Trajopt tutorial

### Backtrace or Console output

Mikael512, 2022-04-01:
how did you set up trajopt? i am trying to install it on ubuntu 20.04 and ros noetic but i cannot simply install openrave 0.9 or openrave 0.8 which is a requirement for trajopt. Do you have any suggestions?

KeerthiSagarSN, 2024-08-29:
@zichunxx Hi, did you by any chance find a working solution for openrave installation in ROS Noetic (Ubuntu 20.04) ?