moveit/moveit_tutorials #659: MoveIt Tutorial for ROS Noetic not working

Wildimp, 2021-08-17:
### Description

I tried to run the second chapter of the MoveIt tutorial for ROS Noetic and cant get past the first command because a urdf/xacro file is missing. I guess it is related to the change of franka_ros from version 0.7.1 to 0.8.0 in which they changed the urdf files.
Changelog of franka_ros:
[https://github.com/frankaemika/franka_ros/blob/noetic-devel/CHANGELOG.md#080---2021-08-03](url)
It was also already adressed in the repo of franka_ros:
[https://github.com/frankaemika/franka_ros/issues/148](url)

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source Build

### Steps to reproduce
Simply follow allong the MoveIt tutorial for ROS Noetic. After finishing the chapter 'Getting started' try to launch the demo.launch file from the second chapter 'MoveIt Quickstart in RViz'. It will throw the following error:

No such file or directory: /opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro [Errno 2] No such file or directory: '/opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro'
RLException: while processing /home/essert/ws_moveit/src/panda_moveit_config/launch/planning_context.launch:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '/opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro']] returned with code [2]. 

Param xml is <param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm_hand.urdf.xacro'"/>
The traceback for the exception was written to the log file

### Expected behaviour
The tutorial for RViz should start

I am pretty new to ROS and github so i am sorry if i am doing mistakes with linking to other sites or if I am saying nonsense.

tammerb, 2021-09-05:
@Wildimp I had the same issue. Follow these pull requests for updates:
https://github.com/ros-planning/panda_moveit_config/pull/83
https://github.com/ros-planning/panda_moveit_config/pull/84

BolunDai0216, 2022-08-30:
You can try to change the following line in `planning_context.launch` of the `panda_moveit_config` package from

```xml
<param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm_hand.urdf.xacro'"/>
```

to 

```xml
<param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm.urdf.xacro' hand:=true"/>
```

rhaschke, 2022-08-30:
This issue was fixed in release 0.7.6

closed 2022-08-30 by rhaschke

CleisonManriqueAguirre, 2023-12-07:
Is not working again

rhaschke, 2023-12-07:
@CleisonManriqueAguirre please open a new issue exactly describing what is not working.