moveit/moveit_tutorials #661: roslaunch panda_moveit_config demo.launch rviz_tutorial:=true is not working

DONJAGA, 2021-08-24:
### I have used moveit_turorials in my Ubuntu LTS 20.4 with ROS noetic without an issue for a while but suddenly occurred an error which stating there is no "/opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro " but when looked for the file it does exists. I even tried to remove and reinstall ROS as well as movit the same error comes finally I removed and reinstalled the entire ubuntu 20.4 LTS but still aster fresh start I'm receiving the save error.   
I have properly sourced the "source ~/ws_moveit/devel/setup.bash" 
<pre>:<font color="#3465A4"><b>~/ws_moveit/src</b></font>$ roslaunch panda_moveit_config demo.launch rviz_tutorial:=true
... logging to /home/don/.ros/log/54180f8c-046f-11ec-8eab-0dbc11f3bdf1/roslaunch-don-39428.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

<font color="#EF2929">No such file or directory: /opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro [Errno 2] No such file or directory: &apos;/opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro&apos;</font>
<font color="#CC0000">RLException: while processing /home/don/ws_moveit/src/panda_moveit_config/launch/planning_context.launch:</font>
<font color="#CC0000">Invalid &lt;param&gt; tag: Cannot load command parameter [robot_description]: command [[&apos;/opt/ros/noetic/lib/xacro/xacro&apos;, &apos;/opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro&apos;]] returned with code [2]. </font>

<font color="#CC0000">Param xml is &lt;param if=&quot;$(eval arg(&apos;load_robot_description&apos;) and arg(&apos;load_gripper&apos;))&quot; name=&quot;$(arg robot_description)&quot; command=&quot;$(find xacro)/xacro &apos;$(find franka_description)/robots/panda_arm_hand.urdf.xacro&apos;&quot;/&gt;</font>
<font color="#CC0000">The traceback for the exception was written to the log file</font>
</pre>
File Avalability 
<pre><font color="#4E9A06"><b>don@don</b></font>:<font color="#3465A4"><b>/opt/ros/noetic/share/franka_description/robots</b></font>$ ls
dual_panda_example.urdf.xacro  hand.urdf.xacro  hand.xacro  panda_arm.urdf.xacro  panda_arm.xacro  panda_gazebo.xacro  utils.xacro
</pre>

PS - my catkin Builds perfectly 
<pre><font color="#4E9A06"><b>don@don</b></font>:<font color="#3465A4"><b>~/ws_moveit/src</b></font>$ catkin build
<font color="#75507B">--------------------------------------------------------</font>
<font color="#06989A">Profile:</font>                     <font color="#C4A000">default</font>
<font color="#06989A">Extending:</font>        <font color="#4E9A06">[explicit]</font> <font color="#C4A000">/opt/ros/noetic</font>
<font color="#06989A">Workspace:</font>                   <font color="#C4A000">/home/don/ws_moveit</font>
<font color="#75507B">--------------------------------------------------------</font>
<font color="#06989A">Build Space:</font>        <font color="#4E9A06">[exists]</font> <font color="#C4A000">/home/don/ws_moveit/build</font>
<font color="#06989A">Devel Space:</font>        <font color="#4E9A06">[exists]</font> <font color="#C4A000">/home/don/ws_moveit/devel</font>
<font color="#06989A">Install Space:</font>      <font color="#3465A4">[unused]</font> <font color="#C4A000">/home/don/ws_moveit/install</font>
<font color="#06989A">Log Space:</font>          <font color="#4E9A06">[exists]</font> <font color="#C4A000">/home/don/ws_moveit/logs</font>
<font color="#06989A">Source Space:</font>       <font color="#4E9A06">[exists]</font> <font color="#C4A000">/home/don/ws_moveit/src</font>
<font color="#06989A">DESTDIR:</font>            <font color="#3465A4">[unused]</font> <font color="#C4A000">None</font>
<font color="#75507B">--------------------------------------------------------</font>
<font color="#06989A">Devel Space Layout:</font>          <font color="#C4A000">linked</font>
<font color="#06989A">Install Space Layout:</font>        <font color="#C4A000">None</font>
<font color="#75507B">--------------------------------------------------------</font>
<font color="#06989A">Additional CMake Args:</font>       <font color="#C4A000">-DCMAKE_BUILD_TYPE=Release</font>
<font color="#06989A">Additional Make Args:</font>        <font color="#C4A000">None</font>
<font color="#06989A">Additional catkin Make Args:</font> <font color="#C4A000">None</font>
<font color="#06989A">Internal Make Job Server:</font>    <font color="#C4A000">True</font>
<font color="#06989A">Cache Job Environments:</font>      <font color="#C4A000">False</font>
<font color="#75507B">--------------------------------------------------------</font>
<font color="#06989A">Whitelisted Packages:</font>        <font color="#C4A000">None</font>
<font color="#06989A">Blacklisted Packages:</font>        <font color="#C4A000">None</font>
<font color="#75507B">--------------------------------------------------------</font>
<font color="#06989A"><b>Workspace configuration appears valid.</b></font>
<font color="#75507B">--------------------------------------------------------</font>
[build] Found &apos;45&apos; packages in 0.0 seconds.                                                                                                                                                               
[build] Package table is up to date.                                                                                                                                                                      
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>geometric_shapes                                              </b></font>                                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>moveit_msgs                                                   </b></font>                                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>moveit_resources_fanuc_description                            </b></font>                                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>moveit_resources_panda_description                            </b></font>                                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>moveit_resources_pr2_description                              </b></font>                                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>moveit_resources_prbt_support                                 </b></font>                                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>rviz_visual_tools                                             </b></font>                                                                                                                              
<font color="#2E3436"><b>Finished</b></font>  <font color="#4E9A06">&lt;&lt;&lt;</font> <font color="#06989A">moveit_resources_panda_description                            </font> [ <font color="#C4A000">0.1 seconds</font> ]                                                                                                              
Starting  <font color="#4E9A06"><b>&gt;&gt;&gt;</b></font> <font color="#06989A"><b>moveit_resources_panda_moveit_config                          </b></font>                                                                                                                              
<font color="#2E3436"><b>Finished</b></font>  <font color="#4E9A06">&lt;&lt;&lt;</font> <font color="#06989A">geometric_shapes                                              </font> [ <font color="#C4A000">0.1 seconds</font> ]                                                                                                              
[... cut ...]
[build]   <font color="#2E3436"><i><b>Abandoned: None.</b></i></font>                                                                                                                                                                                
[build]   <font color="#2E3436"><i><b>Failed:    None.</b></i></font>                                                                                                                                                                                
[build] <i><b>Runtime:</b></i> <i>6.5 seconds total.</i>       </pre>

tylerjw, 2021-08-24:
> `dual_panda_example.urdf.xacro  hand.urdf.xacro  hand.xacro  panda_arm.urdf.xacro  panda_arm.xacro  panda_gazebo.xacro  utils.xacro`

I don't see panda_arm_hand.urdf.xacro in that list.  What does this output?

`ls /opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro`

I think this means franka_description probably changed and released and we need to update something that refers to that file.

DONJAGA, 2021-08-24:
> I don't see panda_arm_hand.urdf.xacro in that list. What does this output?
>
> `ls /opt/ros/noetic/share/franka_description/robots/panda_arm_hand.urdf.xacro`
>
> I think this means franka_description [...]

DONJAGA, 2021-08-24:
> `dual_panda_example.urdf.xacro  hand.urdf.xacro  hand.xacro  panda_arm.urdf.xacro  panda_arm.xacro  panda_gazebo.xacro  utils.xacro`
>
>
> I don't see panda_arm_hand.urdf.xacro in that list. What does [...]

YES, I got it now! Thanks, I thought "panda_arm_hand.urdf.xacro" was there. 
moveit should change their executing script.

closed 2021-08-24 by DONJAGA

tylerjw, 2021-08-24:
> moveit should change their executing script.

Do you mind trying the various other xacro files that are there and letting us know if any of them work well?  If you figure this out you can open a PR to change the file and I'll be happy to help review it.

DONJAGA, 2021-08-29:
The following Solution from users-rizgiak worked for me!!

Remove panda_arm_hand.urdf.xacro changed to panda_arm.urdf.xacro. 

find the file from 

<pre><font color="#4E9A06"><b>don@don</b></font>:<font color="#3465A4"><b>~/ws_moveit/src/panda_moveit_config/launch</b></font>$ ls
chomp_planning_pipeline.launch.xml
default_warehouse_db.launch
demo_chomp.launch
demo.launch
fake_moveit_controller_manager.launch.xml
joystick_control.launch
lerp_planning_pipeline.launch.xml
move_group.launch
moveit_empty.rviz
moveit.rviz
moveit_rviz.launch
ompl-chomp_planning_pipeline.launch.xml
ompl_planning_pipeline.launch.xml
panda_control_moveit_rviz.launch
panda_gripper_moveit_controller_manager.launch.xml
panda_moveit_controller_manager.launch.xml
panda_moveit.launch
panda_moveit_sensor_manager.launch.xml
planning_context.launch
planning_pipeline.launch.xml
run_benchmark_ompl.launch
run_benchmark_trajopt.launch
sensor_manager.launch.xml
setup_assistant.launch
stomp_planning_pipeline.launch.xml
trajectory_execution.launch.xml
trajopt_planning_pipeline.launch.xml
warehouse.launch
warehouse_settings.launch.xml
</pre>

Then use nano or vim to edit the line

//
<pre>  <font color="#4E9A06"><b>don@don</b></font>:<font color="#3465A4"><b>~/ws_moveit/src/panda_moveit_config/launch</b></font>$ sudo nano planning_context.launch  </pre>

//

in the file change line 

    <param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm_hand.urdf.xacro'"/>

to 

    <param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm.urdf.xacro' hand:=true"/>

Then overwrite 

After saving the file use catkin build and launch the moveit using the command. 

<pre>roslaunch panda_moveit_config demo.launch rviz_tutorial:=true</pre>

illusionsoftruth, 2024-09-19:
> The following Solution from users-rizgiak worked for me!!
> 
> Remove panda_arm_hand.urdf.xacro changed to panda_arm.urdf.xacro.
> 
> find the file from
> 
> don@don:~/ws_moveit/src/panda_moveit_config/launch$ ls
> chomp_planning_pipeline.launch.xml
> default_warehouse_db.launch
> demo_chomp.launch
> demo.launch
> fake_moveit_controller_manager.launch.xml
> joystick_control.launch
> lerp_planning_pipeline.launch.xml
> move_group.launch
> moveit_empty.rviz
> moveit.rviz
> moveit_rviz.launch
> ompl-chomp_planning_pipeline.launch.xml
> ompl_planning_pipeline.launch.xml
> panda_control_moveit_rviz.launch
> panda_gripper_moveit_controller_manager.launch.xml
> panda_moveit_controller_manager.launch.xml
> panda_moveit.launch
> panda_moveit_sensor_manager.launch.xml
> planning_context.launch
> planning_pipeline.launch.xml
> run_benchmark_ompl.launch
> run_benchmark_trajopt.launch
> sensor_manager.launch.xml
> setup_assistant.launch
> stomp_planning_pipeline.launch.xml
> trajectory_execution.launch.xml
> trajopt_planning_pipeline.launch.xml
> warehouse.launch
> warehouse_settings.launch.xml
> Then use nano or vim to edit the line
> 
> //
> 
>   don@don:~/ws_moveit/src/panda_moveit_config/launch$ sudo nano planning_context.launch  
> //
> 
> in the file change line
> 
> ```
> <param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm_hand.urdf.xacro'"/>
> ```
> 
> to
> 
> ```
> <param if="$(eval arg('load_robot_description') and arg('load_gripper'))" name="$(arg robot_description)" command="$(find xacro)/xacro '$(find franka_description)/robots/panda_arm.urdf.xacro' hand:=true"/>
> ```
> 
> Then overwrite
> 
> After saving the file use catkin build and launch the moveit using the command.
> 
> roslaunch panda_moveit_config demo.launch rviz_tutorial:=true

I am trying to run `roslaunch panda_sim_moveit sim_move_group.launch ` and I am getting this error :-

RLException: while processing /home/illusionsoftruth/catkin_ws/src/panda_moveit_config/launch/trajectory_execution.launch.xml:
while processing /home/illusionsoftruth/catkin_ws/src/panda_moveit_config/launch/**panda_gripper_moveit_controller_manager.launch.xml**:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: '/home/illusionsoftruth/catkin_ws/src/panda_moveit_config/launch/**panda_gripper_moveit_controller_manager.launch.xml**'
The traceback for the exception was written to the log file

As you can see that it is seraching for **panda_gripper_moveit_controller_manager.launch.xml** but that file no longer exists in this repo anymore.

illusionsoftruth@luciddreamingguy:~/catkin_ws/src/panda_moveit_config/launch$ ls
chomp_planning_pipeline.launch.xml
default_warehouse_db.launch
demo_chomp.launch
demo_gazebo.launch
demo.launch
demo_lerp.launch
demo_stomp.launch
fake_moveit_controller_manager.launch.xml
franka_control.launch
joystick_control.launch
lerp_planning_pipeline.launch.xml
move_group.launch
moveit_empty.rviz
moveit.rviz
moveit_rviz.launch
moveit_scene.rviz
ompl-chomp_planning_pipeline.launch.xml
ompl_planning_pipeline.launch.xml
panda_moveit_sensor_manager.launch.xml
pilz_industrial_motion_planner_planning_pipeline.launch.xml
planning_context.launch
planning_pipeline.launch.xml
ros_controllers.launch
ros_control_moveit_controller_manager.launch.xml
run_benchmark_ompl.launch
run_benchmark_trajopt.launch
sensor_manager.launch.xml
setup_assistant.launch
simple_moveit_controller_manager.launch.xml
stomp_planning_pipeline.launch.xml
trajectory_execution.launch.xml
trajopt_planning_pipeline.launch.xml
warehouse.launch
warehouse_settings.launch.xml

Can someone help me out with this?

illusionsoftruth, 2024-09-19:
Actuallu nevermind, I found it in the kinetic-devel branch