moveit/moveit_tutorials #665: moveit task constuctor tutorial: pick and place demo don't start (Unknown frame: world / Failed to spawn object: table)

robwoidi, 2021-09-15:
### Description

I Try to follow the tutorial for the task constructor. But when I launch the"pickplace.launch" file with "roslaunch moveit_task_constructor_demo pickplace.launch" I get the Error: 

```
terminate called after throwing an instance of std::runtime_error'  what():  Failed to spawn object: table
[mtc_tutorial-1] process has died [pid 28457, exit code -6
```

And in the Terminal, where I first launched the demo file I get the Error 
`[ERROR] [1631713054.898940932]: Unknown frame: world`

### Your environment
ROS Distro: noetic
OS Version: Ubuntu 20.04
Source build

### Steps to reproduce
Just follow the moveIt tutorial for the Task constructor until the step to launch the pickplace.launch file.

### Expected behaviour
The pick and place demo shout start. 

I just started with ROS for my Master thesis so I hope someone of you can help me.

rhaschke, 2021-09-15:
Do you use the latest version of moveit_resources?
See https://github.com/ros-planning/moveit_resources/issues/89 and https://github.com/ros-planning/moveit_resources/issues/97.

closed 2021-09-15 by rhaschke

robwoidi, 2021-09-16:
Thanks! Now it works