moveit/moveit_tutorials #680: Broken link in Gazebo tutorial

rhaschke, 2021-11-10:
The Gazebo tutorial refers to a fork of [`franka_ros` by @tahsinkose](https://github.com/ros-planning/moveit_tutorials/blob/37c070122880dd2da72c05dabec2a4effc826afe/doc/gazebo_simulation/gazebo_simulation.rst#L337-L341). However this fork was obviously deleted recently. @tahsinkose, could you please update the tutorial page correspondingly?

tahsinkose, 2021-11-11:
@rhaschke not only the link but the whole tutorial is broken since `franka_ros` still lacks necessary features.

tahsinkose, 2021-11-11:
@rickstaa could you please provide the exact commits for a working Gazebo+MoveIt integration with the panda robot?

rhaschke, 2021-11-11:
We are in close collaboration with Franka to fix open issues. My WIP branch can be found here: https://github.com/frankaemika/franka_ros/pull/188.
A corresponding `panda_moveit_config` is available on its [noetic-devel branch](https://github.com/ros-planning/panda_moveit_config/tree/noetic-devel).

tahsinkose, 2021-11-11:
@rhaschke I have managed to make integration working with the following two PRs. Most of them are your and @rickstaa's changes.
* https://github.com/rhaschke/franka_ros/pull/2
* https://github.com/ros-planning/panda_moveit_config/pull/98

I'll refresh the tutorial once all development makes their ways to upstreams. Because most of the information in the tutorial is now redundant.

rhaschke, 2021-11-12:
I updated https://github.com/ros-planning/panda_moveit_config/tree/noetic-devel and https://github.com/frankaemika/franka_ros/pull/188 to reflect the latest discussions with Franka. Unfortunately, I'm still missing general feedback for https://github.com/frankaemika/franka_ros/pull/199.
I think both of you, @rickstaa and @tahsinkose, can already start/continue reworking the tutorials based on these commits.
Please try to formulate the tutorials in a generic fashion first (applicable to any robot) and then become more specific to the Panda robot if needed. I tried this, for example [here](https://github.com/ros-planning/moveit_tutorials/pull/678/commits/d5eeb136f46b3d9f08b57f315e7b6c42655b3856#diff-a0586b04b2c8d5cca7eb7878fba3c2d0201ae8f97111e690594a7cdb02e2015eL52-R58).