moveit/moveit_tutorials #681: Please update STOMP planner documentation

MichalDobis, 2021-11-11:
The stomp_core and the stomp_moveit has been split into two separe packages in pull request [882784c](https://github.com/ros-industrial/stomp_ros/commit/882784c6952e6456c85d0f61686913a33309bcfe)

Please, update documentation in https://ros-planning.github.io/moveit_tutorials/doc/stomp_planner/stomp_planner_tutorial.html section Prerequisites step 3. Now both packages are required https://github.com/ros-industrial/stomp_ros and https://github.com/ros-industrial/stomp

rhaschke, 2021-11-12:
You are welcome to file a PR :wink:

MichalDobis, 2021-11-15:
Ok, I suggest this PR: https://github.com/ros-planning/moveit_tutorials/pull/683

mamariomiamo, 2022-02-06:
Number 3 in the prerequisite lists seems to be redundant if the user follow the Installing STOMP from source. Since we need both stomp and stomp_ros packages, I believe it is more accurate to differentiate the name when using wstool set as when I followed the current documentation, it led to build errors.

In fact, stomp_ros's dependencies.rosintall lists the stomp as well as ros_industrial_cke_boilerplate so we can add that into the documentation. I suggest this PR: https://github.com/ros-planning/moveit_tutorials/pull/700

LAbron665, 2022-02-06:
I have the error below when following the current STOMP planner tutorial.
[image]

rhaschke, 2022-02-06:
Addressed in #683.

closed 2022-02-06 by rhaschke