moveit/moveit_tutorials #685: pick_place_tutorial Error

sm3304love, 2021-11-18:
### Description

Overview of your issue here.

When the "pick place tutorial" is executed, the moment the gripper tries to pick up an object, the following error occurs. 

sm3304love@sm3304love-AORUS-5-KB:~/ws_moveit$ rosrun moveit_tutorials pick_place_tutorial 
[ INFO] [1637234143.240412042, 10.037000000]: Loading robot model 'panda'...
[ INFO] [1637234143.240956217, 10.037000000]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1637234144.229894697, 11.026000000]: Ready to take commands for planning group panda_arm.
[ WARN] [1637234152.123228467, 18.913000000]: Fail: ABORTED: Solution found but controller failed during execution
[ WARN] [1637234153.123990351, 19.913000000]: Fail: ABORTED: Must specify group in motion plan request

I create a package via setup_assistant, and followed all the procedures in the following link. What's wrong?
https://github.com/ros-planning/moveit_tutorials/blob/master/doc/gazebo_simulation/gazebo_simulation.rst

This is the git link with the xacro,urdf file I'm currently using.
https://github.com/sm3304love/ros_franka_discreption

### Your environment
* ROS Distro: [melodic]
* OS Version: Ubuntu 18.04

### Expected behaviour
Pick and place tutorial running normally

[image]
[image]

rhaschke, 2021-11-18:
Looks like you didn't configure your controllers correctly.

sm3304love, 2021-11-18:
> Looks like you didn't configure your controllers correctly. What's the content of your `config/*_controllers.yaml`?

Do you mean ros_controllers.yaml?
[image]

sm3304love, 2021-11-18:
`# MoveIt-specific simulation settings
moveit_sim_hw_interface:
    joint_model_group: controllers_initial_group_
    joint_model_group_pose: controllers_initial_pose_
# Settings for ros_control control loop
generic_hw_control_loop:
    loop_hz: 300
    cycle_time_error_threshold: 0.01
# Settings for ros_control hardware interface
hardware_interface:
    joints:
        - panda_joint1
        - panda_joint2
        - panda_joint3
        - panda_joint4
        - panda_joint5
        - panda_joint6
        - panda_joint7
        - panda_finger_joint1
    sim_control_mode: 1  # 0: position, 1: velocity
# Publish all joint states
# Creates the /joint_states topic necessary in ROS
joint_state_controller:
    type: joint_state_controller/JointStateController
    publish_rate: 50
panda_arm_controller:
    type: effort_controllers/JointTrajectoryController
    joints:
        - panda_joint1
        - panda_joint2
        - panda_joint3
        - panda_joint4
        - panda_joint5
        - panda_joint6
        - panda_joint7
    gains:
        panda_joint1: { p: 12000, d: 50, i: 0.0, i_clamp: 10000 }
        panda_joint2: { p: 30000, d: 100, i: 0.02, i_clamp: 10000 }
        panda_joint3: { p: 18000, d: 50, i: 0.01, i_clamp: 1 }
        panda_joint4: { p: 18000, d: 70, i: 0.01, i_clamp: 10000 }
        panda_joint5: { p: 12000, d: 70, i: 0.01, i_clamp: 1 }
        panda_joint6: { p: 7000, d: 50, i: 0.01, i_clamp: 1 }
        panda_joint7: { p: 2000, d: 20, i: 0.0, i_clamp: 1 }

    constraints:
        goal_time: 2.0
    state_publish_rate: 25

panda_hand_controller:
    type: effort_controllers/JointTrajectoryController
    joints:
        - panda_finger_joint1
        - panda_finger_joint2

    gains:
        panda_finger_joint1: { p: 5, d: 3.0, i: 0, i_clamp: 1 }
        panda_finger_joint2: { p: 5, d: 1.0, i: 0, i_clamp: 1 }

    state_publish_rate: 25

controller_list:
    - name: panda_arm_controller
      action_ns: follow_joint_trajectory
      type: FollowJointTrajectory
      default: true
      joints:
        - panda_joint1
        - panda_joint2
        - panda_joint3
        - panda_joint4
        - panda_joint5
        - panda_joint6
        - panda_joint7
    - name: panda_hand_controller
      action_ns: follow_joint_trajectory
      type: FollowJointTrajectory
      default: true
      joints:
        - panda_finger_joint1
        - panda_finger_joint2`

rhaschke, 2021-11-18:
You are using [franka_description's URDF](https://github.com/sm3304love/ros_franka_discreption/blob/main/panda_moveit_config/launch/gazebo.launch#L15).
Which version you are using? The latest version doesn't provide a file `panda_arm_hand.urdf.xacro`?!
Does Gazebo start a `controller_manager`? Check with:
`rosservice list | grep controller_manager`
If not, the URDF is probably missing this tag:
```
    <gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so"/>
    </gazebo>
```

sm3304love, 2021-11-19:
Because the most recent version, Franka_ros, does not provide panda_arm_hand.urdf.xacro, the file was imported from the 0.7.0 version.
When using the  `rosservice list | grep controller_manager` command, ros_controller was not confirmed.

Is it right to add the content to the 'panda_arm_hand.urdf.xacro' file?

rhaschke, 2021-11-19:
> Because the most recent version, franka_ros, does not provide panda_arm_hand.urdf.xacro, the file was imported from 0.7.0.

Don't do this. Instead, apply https://github.com/ros-planning/moveit_tutorials/pull/667.

sm3304love, 2021-11-19:
> > Because the most recent version, franka_ros, does not provide panda_arm_hand.urdf.xacro, the file was imported from 0.7.0.
> 
> Don't do this. Instead, apply #667.

I understood. However, "https://github.com/ros planning/moveit_tutorials/blob/master/doc/gazebo_simulation/gazebo_simulation.rst" explains based on `panda_arm_hand.urdf.xacro`. Is there any explanation for the latest version?

rhaschke, 2021-11-19:
Not yet, but we are working on it. As the new version introduced several regressions, it's not that easy, unfortunately.

sm3304love, 2021-11-19:
Okay. Then I have no choice but to wait. Thank you for your efforts.

sm3304love, 2021-11-19:
Last question. Will it take a long time to complete it?

rhaschke, 2021-11-19:
Just apply the patches from #667 for now and you should be fine.

closed 2021-11-21 by sm3304love