moveit/moveit_tutorials #694: Melodic catkin build

mf093087, 2022-01-16:
_________________________________________________________________________________
Errors     << moveit_tutorials:make /home/user/catkin_3tutorials/logs/moveit_tutorials/build.make.000.log
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp: In function ‘int main(int, char**)’:
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:186:7: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
   req.reference_trajectories.resize(1);
       ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:187:7: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
   req.reference_trajectories[0].joint_trajectory.resize(1);
       ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:189:7: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
   req.reference_trajectories[0].joint_trajectory[0].points.resize(N_STEPS + 1);
       ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:190:7: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
   req.reference_trajectories[0].joint_trajectory[0].joint_names = joint_names;
       ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:191:7: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
   req.reference_trajectories[0].joint_trajectory[0].points[0].positions = current_joint_values;
       ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:203:9: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
     req.reference_trajectories[0].joint_trajectory[0].joint_names = joint_names;
         ^~~~~~~~~~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/trajopt_planner/src/trajopt_example.cpp:204:9: error: ‘planning_interface::MotionPlanRequest {aka struct moveit_msgs::MotionPlanRequest_<std::allocator<void> >}’ has no member named ‘reference_trajectories’
     req.reference_trajectories[0].joint_trajectory[0].points[step_index].positions = joint_values;
         ^~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [doc/trajopt_planner/CMakeFiles/trajopt_example.dir/src/trajopt_example.cpp.o] Error 1
make[1]: *** [doc/trajopt_planner/CMakeFiles/trajopt_example.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp: In function ‘void spawnCollisionObjects(moveit::planning_interface::PlanningSceneInterface&)’:
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:119:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
   box.subframe_names.resize(5);
       ^~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:120:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses.resize(5);
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:122:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
   box.subframe_names[0] = "bottom";
       ^~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:123:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[0].position.y = -.05;
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:124:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[0].position.z = 0.0 + z_offset_box;
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:128:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[0].orientation = tf2::toMsg(orientation);
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:131:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
   box.subframe_names[1] = "top";
       ^~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:132:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[1].position.y = .05;
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:133:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[1].position.z = 0.0 + z_offset_box;
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:135:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[1].orientation = tf2::toMsg(orientation);
       ^~~~~~~~~~~~~~
       plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:137:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_names’
   box.subframe_names[2] = "corner_1";
       ^~~~~~~~~~~~~~
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:138:7: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   box.subframe_poses[2].position.x = -.025;
[... 71 lines cut ...]
   cylinder.subframe_poses[0].position.z = 0.0 + z_offset_cylinder;
            ^~~~~~~~~~~~~~
            plane_poses
/home/user/catkin_3tutorials/src/moveit_tutorials-master/doc/subframes/src/subframes_tutorial.cpp:181:12: error: ‘moveit_msgs::CollisionObject {aka struct moveit_msgs::CollisionObject_<std::allocator<void> >}’ has no member named ‘subframe_poses’; did you mean ‘plane_poses’?
   cylinder.subframe_poses[0].orientation = tf2::toMsg(orientation);
            ^~~~~~~~~~~~~~
            plane_poses
make[2]: *** [doc/subframes/CMakeFiles/subframes_tutorial.dir/src/subframes_tutorial.cpp.o] Error 1
make[1]: *** [doc/subframes/CMakeFiles/subframes_tutorial.dir/all] Error 2
make: *** [all] Error 2
cd /home/user/catkin_3tutorials/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=3,4 -j; cd -

.................................................................................
[... cut ...]
[build] Summary: 4 of 7 packages succeeded.                                      
[build]   Ignored:   None.                                                       
[build]   Warnings:  None.                                                       
[build]   Abandoned: 2 packages were abandoned.                                  
[build]   Failed:    1 packages failed.                                          
[build] Runtime: 32.4 seconds total.                                             
user@ubuntu:~/catkin_3tutorials$

JafarAbdi, 2022-01-17:
Hi @mf093087, could you please provide more info ? which branch/commit are you using ? are you using binaries? if yes what's the moveit version that's you're using?

mf093087, 2022-01-19:
> Hi @mf093087, could you please provide more info ? which branch/commit are you using ? are you using binaries? if yes what's the moveit version that's you're using?

Yes, I am using Moveit melodic version. I solved this issue already by following the official documentation. 
http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html

rhaschke, 2022-01-19:
> I solved this issue already by following the official documentation.

It's always a good idea to carefully read the docs ;-)

closed 2022-01-19 by rhaschke