moveit/moveit_tutorials #698: Position_controllers/JointPositionController is missing from the moveit setup assistance.

sm3304love, 2022-01-22:
### Description

After updating to ubuntu 20.04, I tried to create a new config file using moveit setup assistance. However, position_controllers/JointTrajectoryController was identified in the ROS Control pane selector, but position_controllers/JointPositionController was missing.

### Your environment
* ROS Distro: [noetic]
* OS Version:  Ubuntu 20.04

### Steps to reproduce
https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html

I followed the link above to proceed with the movement setup assistance tutorial.

### Expected behaviour
Select Position_controllers/Joint Position Controller from the ROS Control pane selector.

JafarAbdi, 2022-01-25:
It was removed by https://github.com/ros-planning/moveit/pull/2945/commits/c56d86a4dbf867f09e6392be01b3804ea76343f0, @rhaschke should I update the tutorials to reflect the change ?

sm3304love, 2022-01-25:
> It was removed by [ros-planning/moveit@c56d86a](https://github.com/ros-planning/moveit/commit/c56d86a4dbf867f09e6392be01b3804ea76343f0), @rhaschke should I update the tutorials to reflect the change ?

Yes, currently, the tutorial is explaining based on the old version.

rhaschke, 2022-01-25:
> It was removed by [ros-planning/moveit@c56d86a](https://github.com/ros-planning/moveit/commit/c56d86a4dbf867f09e6392be01b3804ea76343f0), @rhaschke should I update the tutorials to reflect the change?

Yes, it would be great if you could validate and update the tutorial if necessary.
For now, just use the `position_controllers/JointTrajectoryController`.

sm3304love, 2022-01-27:
> > It was removed by [ros-planning/moveit@c56d86a](https://github.com/ros-planning/moveit/commit/c56d86a4dbf867f09e6392be01b3804ea76343f0), @rhaschke should I update the tutorials to reflect the change?
> 
> Yes, it would be great if you could validate and update the tutorial if necessary. @sm3304love, for now, just use the `position_controllers/JointTrajectoryController`.

Thank you. It helped solve the problem.

closed 2022-01-27 by sm3304love