moveit/moveit_tutorials #707: I'm using the ur5_moveit_config package from the universal_robot repository from ROS-Industrial.

rijalmadhav12, 2022-04-11:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rijalmadhav12, 2022-04-11:
**This is my code**
#!/usr/bin/env python3
import sys
import copy
import rospy
import moveit_commander
import moveit_msgs.msg
import geometry_msgs.msg
from math import pi
from std_msgs.msg import String
from moveit_commander.conversions import pose_to_list
moveit_commander.roscpp_initialize(sys.argv)
rospy.init_node('move_group_python_interface_tutorial', anonymous=True)
robot = moveit_commander.RobotCommander()
scene = moveit_commander.PlanningSceneInterface()
group_name = " manipulator"
move_group = moveit_commander.MoveGroupCommander(group_name)
display_trajectory_publisher = rospy.Publisher('/move_group/display_planned_path', moveit_msgs.msg.DisplayTrajectory, queue_size=20)
planning_frame = move_group.get_planning_frame()
joint_goal = move_group.get_current_joint_values()
joint_goal[0] = 0
joint_goal[1] = -pi/4
joint_goal[2] = 0
joint_goal[3] = -pi/2
joint_goal[4] = 0
joint_goal[5] = pi/3
joint_goal[6] = 0
move_group.go(joint_goal, wait=True)
move_group.stop()

**_My gazebo simulation and rviz are perfect.** 
Terminal 1  roslaunch ur_gazebo ur3_bringup.launch
Terminal 2:roslaunch ur3_moveit_config ur3_moveit_planning_execution.launch sim:=true
Terminal 3:roslaunch ur3_moveit_config moveit_rviz.launch config:=true
terminal 4:  python moveplan.py

**I want to plan the motion of ur3 through above python code. When I run above code I get following error_**

[ WARN] [1649716953.644637012]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/robot_description_kinematics/manipulator/kinematics_solver_attempts' from your configuration.
[FATAL] [1649716953.658731962]: Group 'ur3_robot' was not found.
Traceback (most recent call last):
  File "moveplan.py", line 16, in <module>
    move_group = moveit_commander.MoveGroupCommander(group_name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 66, in __init__
    name, robot_description, ns, wait_for_servers
RuntimeError: Group 'ur3_robot' was not found.

**How should I run this python code. Is there a specefic location where it needs to be kept? What would be the group name? I tried "manipulator' and ;endeffector ' as a group name but it didnt work out. Please help**

AzulRadio, 2022-04-17:
open ur3_moveit_config with moveit_setup_assistant and in the part where you define the control groups you should be able to find all the group names. 
ur3_moveit_config/config/ros_controller.yaml might also list them if there is such a file. "endeffector_controller" means the group name is "endeffector" etc.

Also this should be a question for the ROS forum / universal_robot