moveit/moveit_tutorials #709: planning_scene.checkCollision only can check current state?

sunnymints, 2022-04-26:
Moveit demo  give  the example :planning_scene.checkCollision(collision_request, collision_result, copied_state, acm),but the copied_state from   robot_state::RobotState copied_state = planning_scene.getCurrentState(). This demo only can check current state?

but i want to check next state or check all tracks in advance,what can i do？

rhaschke, 2022-04-26:
Just pass all the states you want to check in series.

sunnymints, 2022-04-26:
> Just pass all the states you want to check in series.
Thank you for your reply, can you provide relevant API or functions?  I only know uses the moveit::planning_interface::MoveGroupInterface class to move() one by one, then use planning_scene.checkCollision() . Thank you again.

rhaschke, 2022-04-26:
> maybe this function planning_scene::PlanningScene::setCurrentState

There is no need to change the PlanningScene's RobotState. Just call this variant of `checkCollision`:
```c++
  /** \brief Check whether a specified state (\e robot_state) is in collision. The collision transforms of \e
   * robot_state are
   * expected to be up to date. */
  void checkCollision(const collision_detection::CollisionRequest& req, collision_detection::CollisionResult& res,
                      const moveit::core::RobotState& robot_state) const;
```

sunnymints, 2022-04-27:
Thanks again for your thoughtful reply, I may describe my problem this way. I only know the "moveit_msgs::RobotTrajectory trajectory", But I don't know how to convert"moveit_msgs::RobotTrajectory trajectory" to “moveit::core::RobotState& robot_state” and then send it to ''PlanningScene::checkCollision()''. I want to try your offer the function ''getWayPoint'', but i do not know how to use it(I want a demo).

sunnymints, 2022-04-27:
> I'm not sure what you are missing. Just call `PlanningScene::checkCollision()`, but instead of feeding `PlanningScene::getCurrentState()` you feed the state(s) you want. If you have a `RobotTrajectory` from a planning result, you can access the individual `RobotState` waypoints as follows: [`const moveit::core::RobotState& RobotTrajectory::getWayPoint(std::size_t index) const`](https://github.com/ros-planning/moveit/blob/d19bfac2fac1d8399884928fa29fb0831a6d6a2b/moveit_core/robot_trajectory/include/moveit/robot_trajectory/robot_trajectory.h#L104)

Thanks again for your thoughtful reply, I may describe my problem this way. I only know the "moveit_msgs::RobotTrajectory trajectory", But I don't know how to convert"moveit_msgs::RobotTrajectory trajectory" to “moveit::core::RobotState& robot_state” and then send it to ''PlanningScene::checkCollision()''. I want to try your offer the function ''getWayPoint'', but i do not know how to use it(I want a demo).

rhaschke, 2022-04-27:
Please just read the (API) docs. We cannot provide examples for each and every API function:
http://docs.ros.org/en/noetic/api/moveit_core/html/cpp/classrobot__trajectory_1_1RobotTrajectory.html#aaf22cc512f6d218ab4fafc1eb454d893
There are docs available for each separate MoveIt ROS package: https://moveit.ros.org/documentation/source-code-api

sunnymints, 2022-04-30:
> Please just read the (API) docs. We cannot provide examples for each and every API function: [...]

Thanks again

closed 2022-04-30 by sunnymints