moveit/moveit_tutorials #720: Can't see Panda Robot model in RViz

gdubuque, 2022-07-19:
### Description

Similar to issue [#155 with RViz2](https://github.com/ros-planning/moveit2_tutorials/issues/155), but using RViz and ROS Noetic.

Following the MoveIt Quickstart in RViz tutorial, but the Panda visual model does not appear in RViz. And like the other issue, the collision model can be seen, but not the robot's mesh.

### Your environment
* ROS Distro: Noetic
* OS Version: WIN11, WSL2, Ubuntu 20.04.4

### Steps to reproduce
Followed the [MoveIt Quickstart in RViz tutorial](https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html).
Launched the demo with or without `rviz_tutorial:=true:`
`roslaunch panda_moveit_config demo.launch rviz_tutorial:=true`
`roslaunch panda_moveit_config demo.launch`

After adding the Motion Planning Plugin, and ensuring the Show Robot Visual boxes are checked.

### Expected behaviour
The beautiful Panda robot model in all its glory should be seen in RViz.

:(

MightyMirko, 2022-09-05:
> This is an WSLg issue: [microsoft/wslg#744](https://github.com/microsoft/wslg/issues/744). Try to use software rendering: `LIBGL_ALWAYS_SOFTWARE=true`

I got the same issue but having it virtualized as a vm on proxmox server without gpu passthrough. Your suggestion wasn't the solution :(
ubu2004
ros2 galactic

rhaschke, 2022-09-05:
@MightyMirko, figure out how to render OpenGL stuff in general with your VM.

MightyMirko, 2022-09-07:
@rhaschke 
It should work - glxgears and co. are running good

rhaschke, 2022-09-07:
Maybe, it's the locale problem: https://github.com/ros-visualization/rviz/issues/1151#issuecomment-345726355?

MightyMirko, 2022-09-07:
Uff ... wow - it worked :D

rhaschke, 2022-09-07:
So, you need the locale setting? I just tried with a non-default locale and it still worked for me.

MightyMirko, 2022-09-07:
Hmm i also installed some spice libraries for a problem with virtviewer.. i'll uninstall them and try again 

    sudo apt install 
    mesa-utils
    xserver-xorg-video-qxl
    xserver-xspice
    spice-vdagent

[image]

MightyMirko, 2022-09-07:
yupp... if i set  
    export LC_NUMERIC="de_DE.UTF-8"

it does not show any panda anymore. 

It works with rdp and also with ssh -X

rhaschke, 2022-09-07:
These libs shouldn't affect rviz. 

> If I set `export LC_NUMERIC="de_DE.UTF-8"` it does not show any panda anymore.

It does work for me! Which version of urdfdom do you use?
```
> dpkg -l | grep liburdf
ii  liburdfdom-dev:amd64                                       1.0.4+ds-2                                    amd64        URDF DOM - development files
ii  liburdfdom-headers-dev                                     1.0.4-1                                       amd64        URDF DOM - header files
```

MightyMirko, 2022-09-07:
```

mirko@s25v-ubu2004-ros2-gui:~/move_ws$ dpkg -l | grep liburdf
ii  liburdfdom-dev:amd64                                1.0.4+ds-2                                    amd64        URDF DOM - development files
ii  liburdfdom-headers-dev                              1.0.4-1                                       amd64        URDF DOM - header files
ii  liburdfdom-model:amd64                              1.0.4+ds-2                                    amd64        URDF DOM - model library
ii  liburdfdom-model-state:amd64                        1.0.4+ds-2                                    amd64        URDF DOM - model state library
ii  liburdfdom-sensor:amd64                             1.0.4+ds-2                                    amd64        URDF DOM - sensor library
ii  liburdfdom-world:amd64                              1.0.4+ds-2    
```

MightyMirko, 2022-09-07:
to clarify - i am on moveit2

rhaschke, 2022-09-07:
Please file your issue then over at https://github.com/ros-planning/moveit2_tutorials or https://github.com/ros2/rviz, depending on whether you also don't see the standard rviz RobotModel.

GBW48, 2023-04-18:
I found an answer to the "Can't see Panda Robot model in RViz" problem. I'm currently running the following:
- Windows 11 (10.0.22621.1413), NVIDIA MX550 graphics adapter with latest driver (31.0.15.3141)
- WSL2 (1.1.3.0), VcXsrv (1.20.14.0), Latest NVIDIA Toolkit (12.1)
- Ubuntu 20.04, ROS Noetic, MoveIt 

In addition here is version specific information related to the CUDA, and Mesa graphics environment:

Nvidia-smi
- NVIDIA-SMI 530.41.03, Driver Version  531.41, CUDA Version 12.1, NVIDIA GeForce MX550

Glxinfo | grep “OpenGL version” 
- OpenGL version string: 3.1 Mesa 21.2.6

What follows is a **verified solution**. After the fix, When I run “roslaunch panda_moveit_config demo.launch rviz_tutorial:=true”, I can now see the Panda arm.

After additional research, I found the following post in ROS Answers: [robot-meshes-not-visible-in-rviz-windows11-wsl2](https://answers.ros.org/question/394135/robot-meshes-not-visible-in-rviz-windows11-wsl2/)

In the post user “adrgru” makes the following comment: “Hi, I am using ROS Noetic on Ubuntu 20.04 Windows11 WSL2. When I try to load my robot into rviz or into the MoveIt setup, no meshes are visible. I don't get any errors at all. Here is a picture of rviz( you can only see the two boxes which are the gripper pads):” gripper pads after which it worked as expected, though I did get a lot of warnings regarding the URDF collision defaults.”

Later in the post, user “Sayaki” makes the following observation: “I faced the same issue ( running Noetic + Win11 + WSL2 ). It turns out the problem is with OpenGL. It was using version 3.1. I used a VM to confirm that OpenGL was the culprit, and after updating my mesa-utils by using the instructions given here: [Can wsl2 support a higher version of OpenGL?](https://github.com/microsoft/WSL/discussions/6154) after which it worked as expected, though I did get a lot of warnings regarding the URDF collision defaults.”

With this information and the fact that I was using OpenGl 3.1, I tried following the mesa-utils upgrade instructions, but after two upgrade attempts, I was still using OpenGL 3.1 and Mesa 21.2.6. Not to be deterred, I found another post: [How to Install Mesa Drivers on Ubuntu](https://itsfoss.com/install-mesa-ubuntu/) which discussed how to “Install the latest stable version of Mesa driver in Ubuntu [Latest point release] “.

I used the “Kisak-mesa PPA” repository for the latest point release of Mesa and ran the following terminal commands:

- sudo add-apt-repository ppa:kisak/kisak-mesa
- sudo apt update
- sudo apt upgrade

I then checked the OpenGL and Mesa versions using the glxinfo | grep "OpenGL version" command and got the following results:

- OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.0.2 - kisak-mesa PPA

Since my OpenGL version was now at 4.2, I decided to run “roslaunch panda_moveit_config demo.launch rviz_tutorial:=true”. When RViz opened and I added MotionPlanning to the Display, I saw the Panda arm (when the Planning Group was set to “panda_arm”): [panda_arm](https://drive.google.com/file/d/12JjM6Y2I4hMCCnYE1_P6uX80QnrUkuLu/view?usp=share_link)

When the Planning Group was set to “panda_manipulator”, I saw the Panda manipulator. panda manipulator.
[panda_manipulator](https://drive.google.com/file/d/1LOybonr1j_OkJt4w95fP_bY1tXTFDxmU/view?usp=share_link)

v-xchen-v, 2024-03-20:
try `export LIBGL_ALWAYS_SOFTWARE=true`

DingYoWu, 2024-06-18:
export LIBGL_ALWAYS_SOFTWARE=true 

This command works successfully! Thanks!