moveit/moveit_tutorials #722: IKFast Tutorial: Fix explanation of free index

akgoins, 2022-08-16:
### Description

The IKFast tutorial does not properly explain how to set the `FreeIndex` for a 7DOF robot arm.  It says that the free index is a link, when it actually is a joint.  I have a dual arm robot on a mobile base, and trying to specify a link index always resulted in an error similar to [this post here](https://answers.ros.org/question/285611/set-free_index-for-7-dof-robots-ikfast-moveit-plugin-generation/).  But when I use the joint index, everything works as expected. 

### Steps to reproduce
The problem is that for complex robots, there are a lot of fixed joints which are not indexed by OpenRAVE (`joint_index` and `dof_index` both set to -1.  See below image).  However, all of the links are indexed which means that for a given arm chain the link indices could be in the range (20-26) but the corresponding joint indices could be in the range (10-16).  For my robot the links are 31-37 but the free index is 13 (due to lots of fixed transforms for cameras).

### Expected behaviour
The tutorial needs to be updated to use joints instead of links for the free index.  To find out which joint to specify, use the following command:
```
openrave-robot.py "$MYROBOT_NAME".dae --info joints
```
[image]

Find the joint to set as free (i.e. panda_joint1) and the `joint_index` for that joint.  Then set the environment variable `FREE_INDEX=0` and then run `ikfast.py` as before.

rhaschke, 2022-08-16:
Thanks for reporting this issue. Do you mind filing a PR as well?

rhaschke, 2022-09-02:
I guess you were referring to the [Kinetic tutorial](http://docs.ros.org/en/kinetic/api/moveit_tutorials/html/doc/ikfast/ikfast_tutorial.html)? See #725 for the proposed new wording.

rhaschke, 2022-09-02:
Fixed via #725

PR #725 (2022-09-02): IKFast: Fix description of free link -> free joint
Fixes #722
closed 2022-09-02 by rhaschke