moveit/moveit_tutorials #724: Error when running c++ tutorial

BolunDai0216, 2022-08-30:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source or Binary build? Binary, MoveIt1 Noetic

### Steps to reproduce
I was simply running 

```console
roslaunch panda_moveit_config demo.launch
```
and in another terminal

```console
roslaunch moveit_tutorials move_group_interface_tutorial.launch
```

### Expected behaviour
RViz looks like this:

[image]

The following error occurs:

```console
[move_group-5] process has died [pid 760183, exit code -11, cmd /home/bolun/ws_moveit/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/bolun/.ros/log/676c8b16-287d-11ed-805e-97f6181ad492/move_group-5.log].
log file: /home/bolun/.ros/log/676c8b16-287d-11ed-805e-97f6181ad492/move_group-5*.log
```

### Backtrace or Console output

The console output of `roslaunch panda_moveit_config demo.launch` can be found [here](https://gist.github.com/BolunDai0216/09eccfaee7e8079ac14c44afd4abec54). The console output of `oslaunch moveit_tutorials move_group_interface_tutorial.launch` can be found [here](https://gist.github.com/BolunDai0216/f776a426dac40f99a14ff67e636b380b)

BolunDai0216, 2022-09-01:
> Hi @BolunDai0216,
> 
> Looks like your move_group node is segfaulting, do you mind sharing the stacktrace when that happens?

Do you know how I can get the stacktrace? All I got from the console is what I shared in the `Backtrace or Console output` section of my question.

tylerjw, 2022-09-01:
Here is the wiki page on lauching in gdb for ros1: http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB