moveit/moveit_tutorials #738: Realtime Arm Servoing tutorial obsolete

slabua, 2022-11-15:
### Description

As per title.
Various commands do not match the latest status of the ros-industrial/universal_robot repository.

### Your environment
* ROS Distro: Noetic

rhaschke, 2022-11-15:
I guess you mean the tutorial is outdated, not obsolete?
Please provide a PR to fix the broken commands. Thanks.

Robotawi, 2022-11-17:
I encountered some problems with the servo tutorial before and solved them. I will follow the tutorial again and suggest modifications in a PR accordingly. Feel free to assign this to me @rhaschke

slabua, 2022-11-17:
@Robotawi thank you! Have you also used it with a real robot? I'm trying to move an UR5e and I'm half way through.

Robotawi, 2022-11-18:
My pleasure! It was a Gazebo-simulated xArm7 robot @slabua 
I will make sure to try on a real robot this time.

It is great you made some progress. Can you describe the issues you encountered?

slabua, 2022-11-18:
Hi @Robotawi I ended up using the kinetic branch of ros-industrial/universal_robot with the remaining stuff set for noetic since I did not have enough time to find out the new commands and controller names. That is not ideal but two obsolete things worked well together after all.
Also, in order to use the real robot I had to use packages from the Universal_Robots_ROS_Driver repository and run the external control urcap on the pendant.
Furthermore, I had to use the velocity controller instead of the position controller of moveit_servo.
Today it moved by publishing data on the twist topic, that was enough to go back home with peace of mind.

rhaschke, 2022-11-21:
Thanks, @Akumar201 for this info. It would be great if you could file a PR providing the newly required commands.
You can edit and file a PR directly from your browser: https://github.com/ros-planning/moveit_tutorials/edit/master/doc/realtime_servo/realtime_servo_tutorial.rst

Akumar201, 2022-11-30:
Hi, I found out that right now the easiest way would be to use the old version from https://github.com/ros-industrial/universal_robot/tags 1.2.7. I am working on the latest version to make it compatible and will push the changes once done.

Akumar201, 2023-01-30:
@rhaschke I made the changes, please verify .

rhaschke, 2023-01-30:
> @rhaschke I made the changes, please verify.

Thanks! But, I can't find your pull request.

Akumar201, 2023-01-30:
Hi, can you see it now?