moveit/moveit_tutorials #740: speed up the display of the trajectory/path in rviz

chauvinSimon, 2022-12-09:
### Description

I would like to ask **if it is possible to speed up the display of the planned path** in rviz.

### motivation
- I would like to **see the planned path in rviz ahead of the current robot**
- obviously I could wait for an `input()` as in the [moveit tutorial](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html), asking the user to press `Enter` to start the display, and another one later to start the execution.
- or even add some sleep before the execution

### example

here the planned path is displayed **just a bit ahead of the real motion**. I would be nice to have it running faster so that the planned states appear more ahead.

[image]

### what I already tried
In the [DisplayTrajectory type](https://docs.ros.org/en/noetic/api/moveit_msgs/html/msg/DisplayTrajectory.html) there is no field for such a display speed and I could not find any setting in the moveit plugin in rviz (see current settings in the above gif).

### to reproduce

I have adapted the [python tutorial](https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html#displaying-a-trajectory) to display the trajectory

```
        self.display_trajectory_publisher = rospy.Publisher(
            "/move_group/display_planned_path",
            moveit_msgs.msg.DisplayTrajectory,
            queue_size=20,
        )
        ....

        # display a saved trajectory
        display_trajectory = moveit_msgs.msg.DisplayTrajectory()
        display_trajectory.trajectory_start = self.robot.get_current_state()
        display_trajectory.trajectory.append(plan)
        self.display_trajectory_publisher.publish(display_trajectory)

        # I could add a sleep here. but that is not nice

        success = self.move_group.execute(plan, wait=True)
```

### my environment

ROS Distro: [noetic]
OS Version: e.g. Ubuntu 20.04

rhaschke, 2022-12-09:
Just modify `State Display Time` of the `DisplayTrajectory` plugin!

chauvinSimon, 2022-12-09:
Thank you very much @rhaschke !

> State Display Time: Replay speed of trajectory. Either as factor of its timeparameterization or as constant time (s) per waypoint.

how could I have missed that?! :see_no_evil:

closed 2022-12-09 by chauvinSimon