moveit/moveit_tutorials #742: Moveit task constructor cannot find the end effector

juandavid212, 2023-01-04:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2023-01-04:
You need to define an end-effector in your robot's `.srdf` file and refer to that eef in the config file:
https://github.com/ros-planning/moveit_task_constructor/blob/master/demo/config/panda_config.yaml#L6

juandavid212, 2023-01-23:
Thanks for the response. It worked.
I'm now having trouble defining the pre-grasp Pose (stage->setPreGraspPose(hand_open_pose_)). 
I tried defining an empty group_state on the gripper at hand.xacro file like 
<group_state name="open" group="hand"></group_state>
 But I still get the following error:
Initialization failed: Error initializing stage:
generate grasp pose: invalid pregrasp: unknown state 'open

My end effector is always open, and I would like to position it relative to a certain pose/object (cylinder). Similar to the pick-and-place demo but without grasping.

rhaschke, 2023-01-23:
Opening and closing the fingers before/during grasping is an inherent part of the pick task.
If you want to skip that, you should plan just a simple trajectory to a specific target given relative to your object.

juandavid212, 2023-01-24:
I see, I would need something that allows me to spawn poses with an 'angle_delta' interval around a specific frame or point. I think a simple trajectory doesn't allow me to do this.

rhaschke, 2023-01-24:
I suggest modifying the existing Pick+Place example to your needs, i.e. removing Grasp+Place stuff and just keep the (grasp) pose generator.

rhaschke, 2023-01-30:
> How can I execute the plan (solution of MTC) in moveit?

Just call `task.publish(task.solutions[0])` in Python or `task->execute(*task->solutions().front())` in C++.
You can also trigger execution from rviz by pressing the corresponding button.

rhaschke, 2023-01-30:
> Would you happen to know the function for C++?

It was stated in https://github.com/ros-planning/moveit_tutorials/issues/742#issuecomment-1408470610: `task->execute(*task->solutions().front())`

juandavid212, 2023-01-30:
I see,
It seems that the code get's stuck in "Executing solution trajectory." I receive a warning at the simulation command line, " Empty quaternion found in pose message. Setting to neutral orientation." and nothing happens.
This is the code I have

`
#include <test_plan/test_class.h>
#include <rosparam_shortcuts/rosparam_shortcuts.h>
#include <actionlib/client/simple_action_client.h>

namespace test_plan {

constexpr char LOGNAME[] = "test_plan";
constexpr char PickPlaceTask::LOGNAME[];

void spawnObject(moveit::planning_interface::PlanningSceneInterface& psi, const moveit_msgs::CollisionObject& object) {
	if (!psi.applyCollisionObject(object))
		throw std::runtime_error("Failed to spawn object: " + object.id);
}

moveit_msgs::CollisionObject createObject(ros::NodeHandle& pnh) {
	std::string object_name, object_reference_frame;
	std::vector<double> object_dimensions;
	geometry_msgs::Pose pose;
	std::size_t error = 0;
	error += !rosparam_shortcuts::get(LOGNAME, pnh, "object_name", object_name);
	error += !rosparam_shortcuts::get(LOGNAME, pnh, "object_reference_frame", object_reference_frame);
	error += !rosparam_shortcuts::get(LOGNAME, pnh, "object_dimensions", object_dimensions);
	error += !rosparam_shortcuts::get(LOGNAME, pnh, "object_pose", pose);
	rosparam_shortcuts::shutdownIfError(LOGNAME, error);

	moveit_msgs::CollisionObject object;
	object.id = object_name;
	object.header.frame_id = object_reference_frame;
	object.primitives.resize(1);
	object.primitives[0].type = shape_msgs::SolidPrimitive::CYLINDER;
	object.primitives[0].dimensions = object_dimensions;
	pose.position.z += 0.5 * object_dimensions[0];
	object.primitive_poses.push_back(pose);
	return object;
}

void setupDemoScene(ros::NodeHandle& pnh) {
	// Add table and object to planning scene
	ros::Duration(1.0).sleep();  // Wait for ApplyPlanningScene service
	moveit::planning_interface::PlanningSceneInterface psi;
	spawnObject(psi, createObject(pnh));
}

PickPlaceTask::PickPlaceTask(const std::string& task_name, const ros::NodeHandle& pnh)
  : pnh_(pnh), task_name_(task_name) {
	loadParameters();
}

void PickPlaceTask::loadParameters() {
	/****************************************************
	 *                                                  *
	 *               Load Parameters                    *
	 *                                                  *
	 ***************************************************/
	ROS_INFO_NAMED(LOGNAME, "Loading task parameters");

	// Planning group properties
	size_t errors = 0;
	errors += !rosparam_shortcuts::get(LOGNAME, pnh_, "arm_group_name", arm_group_name_);
	errors += !rosparam_shortcuts::get(LOGNAME, pnh_, "hand_group_name", hand_group_name_);
	errors += !rosparam_shortcuts::get(LOGNAME, pnh_, "eef_name", eef_name_);
	errors += !rosparam_shortcuts::get(LOGNAME, pnh_, "hand_frame", hand_frame_);
	errors += !rosparam_shortcuts::get(LOGNAME, pnh_, "grasp_frame_transform", grasp_frame_transform_);

	// Predefined pose targets
[... 163 lines cut ...]
}

bool PickPlaceTask::execute() {
	ROS_INFO_NAMED(LOGNAME, "Executing solution trajectory");
	moveit_msgs::MoveItErrorCodes execute_result;

	execute_result = task_->execute(*task_->solutions().front());
	// // If you want to inspect the goal message, use this instead:
	// actionlib::SimpleActionClient<moveit_task_constructor_msgs::ExecuteTaskSolutionAction>
	// execute("execute_task_solution", true); 
	// execute.waitForServer();
	// moveit_task_constructor_msgs::ExecuteTaskSolutionGoal execute_goal;
	// task_->solutions().front()->fillMessage(execute_goal.solution);
	// execute.sendGoalAndWait(execute_goal);
	// execute_result = execute.getResult()->error_code;

	if (execute_result.val != moveit_msgs::MoveItErrorCodes::SUCCESS) {
		ROS_ERROR_STREAM_NAMED(LOGNAME, "Task execution failed and returned: " << execute_result.val);
		return false;
	}

	return true;
}
}  // namespace moveit_task_constructor_demo
`

rhaschke, 2023-01-30:
Execution is not related to MTC. Thus, your code doesn't help to diagnose a potential issue.
Have a look at your MoveIt and/or Gazebo console!
The warning `Empty quaternion found in pose message. Setting to neutral orientation.` probably originates from rviz and isn't pointing into the right direction either.

rhaschke, 2023-01-31:
You need to run your `move_group` node and configure the `move_group` capability by adding the following line to your `move_group.launch`:
```xml
<param name="move_group/capabilities" value="move_group/ExecuteTaskSolutionCapability" />
```
Which documentation did you check?

juandavid212, 2023-02-03:
Thanks for the answer.

I checked this documentation(https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html)

I'm trying to move the panda's end effector(the same robot as the demo) to a specific pose using C++; I've found this example in python (https://ros-planning.github.io/moveit_task_constructor/howto.html#generatepose), and I'm trying to do a C++ version. The code runs fine, and the plan is published to RVIZ, but the end effector does not reach the desired pose. It doesn't move.

My code currently looks like this:

	auto sampling_planner = std::make_shared<solvers::PipelinePlanner>();
	sampling_planner->setProperty("goal_joint_tolerance", 1e-5);

	// auto pp{ std::make_shared<solvers::PipelinePlanner>("ompl") };
	// pp->setPlannerId("RRTConnectkConfigDefault");

	Task t;
	t.stages()->setName("Move_To_Pose");
	//t.setProperty("ik_frame", "panda_hand");
	Stage* current_state_ptr = nullptr;  // Forward current_state on to grasp pose generator
	auto current_state = std::make_unique<stages::CurrentState>("current");

	current_state_ptr = current_state.get();
	t.add(std::move(current_state));

	{  // Move-t
		auto stage = std::make_unique<stages::Connect>(
		    "connect 1", stages::Connect::GroupPlannerVector{ { "panda_arm", sampling_planner } });
		stage->setTimeout(5.0);
		stage->properties().configureInitFrom(Stage::PARENT);
		t.add(std::move(stage));
	}

	{
		auto stage = std::make_unique<stages::GeneratePose>("moveTo");
		//stage->properties().configureInitFrom(Stage::PARENT);
		//stage->properties().set("link", "panda_hand");
		geometry_msgs::PoseStamped pose_msg;
		pose_msg.header.frame_id = "panda_hand";
		pose_msg.pose.position.x = 0.5;
		pose_msg.pose.position.y = 0.5;
		pose_msg.pose.position.z = 0.5;
		pose_msg.pose.orientation.x = 0;
		pose_msg.pose.orientation.y= 0;
		pose_msg.pose.orientation.z = 0;
		pose_msg.pose.orientation.w = 1;

		stage->setPose(pose_msg);
		stage->setMonitoredStage(current_state_ptr);  // Hook into attach_object_stage
		t.add(std::move(stage));
	}	

And the output of my console is as follows:

 INFO] [1675441162.135474077]: Loading robot model 'panda'...
[ INFO] [1675441162.179179940]: Using planning interface 'OMPL'
[ INFO] [1675441162.180303177]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1675441162.180501865]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1675441162.180650475]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1675441162.180801187]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1675441162.180938689]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1675441162.181073228]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1675441162.181083654]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1675441162.181089472]: Using planning request adapter 'Resolve constraint frames to robot links'
[ INFO] [1675441162.181094069]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1675441162.181098563]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1675441162.181103175]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1675441162.181107462]: Using planning request adapter 'Fix Start State Path Constraints'
[ INFO] [1675441162.188009664]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1675441162.188198293]: panda_arm/panda_arm: Starting planning with 1 states already in datastructure
[ INFO] [1675441162.198745250]: panda_arm/panda_arm: Created 5 states (2 start + 3 goal)
[ INFO] [1675441162.198750962]: Solution found in 0.010598 seconds
[ INFO] [1675441162.200263185]: SimpleSetup: Path simplification took 0.001494 seconds and changed from 4 to 2 states
  1  - ←   1 →   -  1 / Move_To_Pose
    1  - ←   1 →   -  1 / current
    -  1 →   1 ←   1  - / connect 1
    1  - ←   1 →   -  1 / moveTo

Thanks

closed 2023-03-14 by juandavid212