moveit/moveit_tutorials #747: Invalid Trajectory: start point deviates from current robot state more than 0.01

aubino, 2023-01-25:
This line in the move group tutorial causes the error on this topic.
It actually tries to  separately modify the current state while not moving resulting in the error above.   I don't know if it is worth a pull request though. 
https://github.com/ros-planning/moveit_tutorials/blob/adbd6cc012c5b8f647803494189f9d8654a12935/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L134

rhaschke, 2023-01-25:
This section of the tutorial illustrates planning with path constraints. To this end, the start state needs to satisfy the constraints:
https://github.com/ros-planning/moveit_tutorials/blob/adbd6cc012c5b8f647803494189f9d8654a12935/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L241-L243

So, it is desired to start from a virtual start state. But: the tutorial doesn't execute the trajectory. How do you get the execution error then? Did you execute the trajectory yourself?

aubino, 2023-01-25:
> This section of the tutorial illustrates planning with path constraints. To this end, the start state needs to satisfy the constraints: [...]

closed 2023-01-25 by aubino