moveit/moveit_tutorials #751: How to use velocity control in MoveIt?

Akumar201, 2023-01-30:
### Description
Hi, I would like to know how can I use velocity control in Moveit when planning [Cartesian Path ](http://docs.ros.org/en/indigo/api/moveit_tutorials/html/doc/pr2_tutorials/planning/src/doc/move_group_interface_tutorial.html). Is there any specific thing I need to do in order to move my robot according to velocity control

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 22.04

rhaschke, 2023-01-31:
Just configure a velocity-based `JointTrajectoryController` in your `controllers.yaml`.

Akumar201, 2023-02-01:
I am closing this issue as it has been solved. If the problem persists, please comment and the issue will be reopened if appropriate.

closed 2023-02-01 by Akumar201