moveit/moveit_tutorials #755: Catkin Build Error on Jetson

ghimireadarsh, 2023-02-07:
### Description
I am trying to install the moveit tutorial but i fails on the "catkin build" command.

Overview of your issue here.
After using the `catkin build -j 1` command

Overview of your issue here.
Just following steps here -> [From getting started](http://docs.ros.org/melodic/api/moveit_tutorials/html/doc/getting_started/getting_started.html) following the previous issues resolved case using `-j 1` or `--jobs 1` tag in catkin build and it fails.

### Your environment
* ROS Distro: Melodic
* OS Version: Ubuntu 18.04.6 LTS
* gcc+7

### Output
```shell
Workspace configuration appears valid.
-----------------------------------------------------------
[build] Found '2' packages in 0.0 seconds.                                     
[build] Package table is up to date.                                           
Starting  >>> panda_moveit_config                                              
Finished  <<< panda_moveit_config                [ 0.6 seconds ]               
Starting  >>> moveit_tutorials                                                 
_______________________________________________________________________________
Errors     << moveit_tutorials:make /home/nvidia/ws_moveit/logs/moveit_tutorials/build.make.009.log
c++: internal compiler error: Segmentation fault (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [doc/motion_planning_api/CMakeFiles/motion_planning_api_tutorial.dir/src/motion_planning_api_tutorial.cpp.o] Error 4
make[1]: *** [doc/motion_planning_api/CMakeFiles/motion_planning_api_tutorial.dir/all] Error 2
make: *** [all] Error 2
cd /home/nvidia/ws_moveit/build/moveit_tutorials; catkin build --get-env moveit_tutorials | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << moveit_tutorials:make              [ Exited with code 2 ]        
Failed    <<< moveit_tutorials                   [ 25.2 seconds ]              
[build] Summary: 1 of 2 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 25.9 seconds total.
```

rhaschke, 2023-02-07:
Typically we see such compiler crashes on machines with too little memory. Please check your memory consumption!

ghimireadarsh, 2023-02-08:
I checked my memory consumption, and it looked fine. As I thought of that could have been the reason, so I also increased the Swap memory size, which seemed to not have been used either.

rhaschke, 2023-02-08:
I'm afraid we cannot help any further with that issue. Obviously, gcc for ARM is broken on your machine. Can you update to a newer version of gcc in hope that the bug was already fixed?

ghimireadarsh, 2023-02-10:
Thank you for your response. I have been trying change the gcc to newer version, looks like not an easy job for arm machine.

closed 2023-02-10 by ghimireadarsh