moveit/moveit_tutorials #757: Cannot add again a deleted colision object.

juandavid212, 2023-02-09:
Description

Overview of your issue here:
I'm trying to add a collision object, remove it, and added again in a different location. But I'm not able to. I receive the following error when adding the object:
 More primitive shape poses than shapes in collision object message.
Your environment

    ROS Distro: noetic
    OS Version: Ubuntu 20.04

Steps to reproduce
I'm creating the object like this:

 geometry_msgs::Pose p;
  p.position.x = 0.2;
  p.position.y = 0.2;
  p.position.z = 0.2;
  p.orientation.x = 0;
  p.orientation.y = 0;
  p.orientation.z = 0;
  p.orientation.w = 1;

  object_.id = "cane";
  object_.header.frame_id = base_link;
  object_.primitives.resize(1);
  object_.primitives[0].type = shape_msgs::SolidPrimitive::CYLINDER;
  object_.primitives[0].dimensions = {0.01, 0.001};
  object_.primitive_poses.push_back(p);

  object_.operation = object_.ADD;
  
  ps_.world.collision_objects.push_back(object_);
  ps_.is_diff = true;
  psd_publisher_.publish(ps_);

And I'm removing it using this:

  object_.operation = object_.REMOVE;

  ROS_INFO("Removing the object from the world.");
  ps_.robot_state.attached_collision_objects.clear();
  ps_.world.collision_objects.clear();
  ps_.world.collision_objects.push_back(object_);
  ps_.is_diff = true;
  psd_publisher_.publish(ps_);

After that, I check with:
rosrun moveit_ros_planning moveit_print_planning_scene_info
and I'm able to see when the object is added and removed. 
But when I'm trying to add the object after deleting it I receive the error

More primitive shape poses than shapes in collision object message.

Tell us what should happen
I expect to add the removed object even if it has the same name and pose.

Use [gist.github.com](https://github.com/ros-planning/moveit_tutorials/issues/gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2023-02-09:
The error message exactly describes your issue: You have more `primitive_poses` (probably 2) than `primitives` (only 1), because you `push_back`, i.e. accumulate poses.

juandavid212, 2023-02-10:
Thanks, 
I've putted 
object_.primitive_poses.clear();
before pushing the pose.

closed 2023-02-10 by juandavid212