moveit/moveit_tutorials #767: Confusion about IKFast installation

bjutlss, 2023-04-04:
### Description

I tried to configure the IKFast plugin for my 5D robotic arm in the `ros-noetic` version and installed it according to the IKFast Kinematics Solver, completing all the steps, including installing the docker image `personalrobotics/ros-openrave`. But when running the command `roslaunch "$MYROBOT_NAME"_moveit_config demo.launch` to verify the feasibility, I can't drag the interactive ball at the end.
I'm confused about this, is it because I didn't generate a container based on the image that caused ikfast to be unable to call openrave? Or something else, I hope I can get an answer.

### Your environment
* ROS Distro: Noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? Source

### Steps to reproduce
I followed the instructions step by step to install. 
https://ros-planning.github.io/moveit_tutorials/doc/ikfast/ikfast_tutorial.html

### Backtrace or Console output
```
********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
*     - SequenceAction
*     - SequenceService
********************************************************

[ INFO] [1680603236.727396100]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1680603236.727408857]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1680603236.846412557]: Stereo is NOT SUPPORTED
[ INFO] [1680603236.846452055]: OpenGL device: NVIDIA GeForce RTX 3070 Ti/PCIe/SSE2
[ INFO] [1680603236.846462993]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1680603240.111642052]: Loading robot model 'robot_model'...
[ WARN] [1680603240.150352061]: Could not identify parent group for end-effector 'gripper'
[ INFO] [1680603240.156224530]: Using empty link_prefix.
[ INFO] [1680603240.303807502]: Starting planning scene monitor
[ INFO] [1680603240.306669220]: Listening to '/move_group/monitored_planning_scene'
[ WARN] [1680603240.371524813]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1680603240.374970111]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1680603241.574876300]: Ready to take commands for planning group manipulator.

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.
```

bjutlss, 2023-04-05:
Sorry for not replying in time. I cannot find the log of the generator script auto_create_ikfast_moveit_plugin.sh,Can you give me some tips on logs? For example the exact path.
Thanks.

rhaschke, 2023-04-05:
I'm afraid there is no log file of this, but you need to repeat the generation process.

bjutlss, 2023-04-05:
In fact, I have followed the instructions and installed it no less than 5 times, for which I have tried to configure 6D ikfast for ur5 and it is working fine. But configuring ikfast with TranslationDirection5Dd for my own 5-DOF robot arm does not work properly.

bjutlss, 2023-04-05:
> I'm sorry for that, but without the log, we cannot help.
Where should the log file be generated? Is it a normal phenomenon to have no log files? I tried searching the entire workspace for a file like `ikfast.log`, but no luck. Can you tell me the name of the log file?

bjutlss, 2023-04-05:
I might see what you mean, I will try to generate my log file. Thanks for your reply!

Nikerane, 2025-01-12:
I have a similar confusion when generating the plugin for mycustom 4DoF robot as after generating it gives openravepy.databases.inversekinematics: testik, success rate: 0.000000, wrong solutions: 0.000000, no solutions: 1.000000, missing solution: 0.000000.
Added the log and the Arm robot file. It would be great if someone could suggest any ideas to address this

[4DoF_ikfast_success0.txt](https://github.com/user-attachments/files/18388959/4DoF_ikfast_success0.txt)

[arm_robot.zip](https://github.com/user-attachments/files/18388962/arm_robot.zip)

rhaschke, 2025-01-12:
If you aim for `--iktype Transform6D`, a 4DoF or 5DoF arm obviously cannot reach all 6D poses.