moveit/moveit_tutorials #775: I am getting the folllowing issue for moveit_ros_planning_interface.

Kishor-Ryouta, 2023-04-15:
CMake Error at /home/kishor/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Python3 (missing: NumPy Development Development.Module
  Development.Embed) (found version "3.8.10")
Call Stack (most recent call first):
  /home/kishor/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  /home/kishor/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.26/Modules/FindPython/Support.cmake:3766 (find_package_handle_standard_args)
  /home/kishor/.local/lib/python3.8/site-packages/cmake/data/share/cmake-3.26/Modules/FindPython3.cmake:551 (include)
  /opt/ros/noetic/share/eigenpy/cmake/python.cmake:170 (find_package)
  /opt/ros/noetic/share/eigenpy/cmake/eigenpyConfig.cmake:144 (FINDPYTHON)
  CMakeLists.txt:41 (find_package)

rhaschke, 2023-04-15:
You are using a custom cmake version (3.26 installed in ~/.local/lib/python3.8/site-packages/cmake).
The default for Focal/Noetic is cmake 3.16.3 from /usr/bin/cmake. Are you using a python virtualenv?

Kishor-Ryouta, 2023-04-15:
I am using windows WSL2 btw.  what do you mean by python virtual environment? (sorry, i am not well versed with the terminologies as I am from a hardware background)

Plus, the default version for cmake is 3.26 when I completely ireinstall cmake.

Kishor-Ryouta, 2023-04-15:
it builds upto 29 packages post which catkin fails to build 

[image]

Kishor-Ryouta, 2023-04-15:
[image]
sharing the error messages image as well just in case

rhaschke, 2023-04-15:
How did you install cmake? I guess via pip. That's not the recommended way to manage a ROS system.
Please follow the [install instructions](https://moveit.ros.org/install/source), which suggests `rosdep` to install required dependencies.

rhaschke, 2023-04-15:
Probably your pip-installed cmake still works, but you are missing numpy now. Either try `pip install numpy` or use the recommended installation process.

Kishor-Ryouta, 2023-04-15:
i used sudo apt install cmake

and I already have numpy installed.
i even tried spcifiying  the python3 path as flag to 'catkin_make'. but the error still occurs

Kishor-Ryouta, 2023-04-15:
> How did you install cmake? I guess via pip. That's not the recommended way to manage a ROS system. Please follow the [install instructions](https://moveit.ros.org/install/source), which suggests `rosdep` to install required dependencies.

i used the same link, yet the build failes at 'moveit_ros_planning_interface'

rhaschke, 2023-04-15:
Probably you installed cmake via pip in the past then. 

> I used `sudo apt install cmake`

Where is cmake found? What is the result of `which cmake`? I guess, `~/.local/bin/cmake`, which is wrong.

Obviously, you screwed your Ubuntu installation. If you are not able to recover, I suggest starting over and installing Ubuntu from scratch. Closing here as off-topic.

closed 2023-04-15 by rhaschke

Kishor-Ryouta, 2023-04-15:
how do I install cmake over again?
and what should be the result of 'which cmake'?

simonschmeisser, 2023-04-16:
It should start with `/usr/`

You can also check that your `PATH` and `PYTHONPATH` do not contain `.local`