moveit/moveit_tutorials #779: CI broken: htmlproofer reports spurious error

rhaschke, 2023-05-20:
Since a few days `htmlproofer` is broken and reports this spurious error in [`realtime_servo_tutorial.rst`](https://github.com/ros-planning/moveit_tutorials/blob/9ad1fecf00f6f40588eb5c88508347f00affdd6c/doc/realtime_servo/realtime_servo_tutorial.rst?plain=1#L8-L15):
  ```
    *  877:3: ERROR: Eng tag 'embed' isn't allowed here. Currently open tags: html, body, div, section, div, div, div, div, div, object.
    </embed>
  ```
Obviously, the corresponding source code is ok:
https://github.com/ros-planning/moveit_tutorials/blob/9ad1fecf00f6f40588eb5c88508347f00affdd6c/doc/realtime_servo/realtime_servo_tutorial.rst?plain=1#L8-L15
The error was introduced by external changes or updates as the same commit that [built successfully 5 days ago](https://github.com/ros-planning/moveit_tutorials/actions/runs/4977228410/attempts/1), [now fails](https://github.com/ros-planning/moveit_tutorials/actions/runs/4977228410/attempts/2): https://github.com/ros-planning/moveit_tutorials/actions/runs/4977228410

I don't have time to tackle this. Hopefully, somebody from your team, @davetcoleman / @henningkayser, can have a look.
I suggest updating the workflow using the template from moveit2_tutorials.

Note that the htmlproofer job on `moveit2_tutorials` is [broken too](https://github.com/ros-planning/moveit2_tutorials/actions/runs/4997935657) - but for different reasons.

henningkayser, 2023-05-20:
We have seen something similar on the MoveIt website. https://github.com/ros-planning/moveit.ros.org/issues/803
And there is actually a bunch of broken html, partially in the layouts and partially due to markdown processing, so it seems. I didn't have too much time to look into it yet, but it's certainly reproducible locally and not really spurious at all.

rhaschke, 2023-05-20:
I have seen the MoveIt2 issue, but it is unrelated. Here, the error reported by `htmlproofer` is definitely spurious. The generated HTML is perfectly fine.

henningkayser, 2023-05-22:
The HTML is not really fine, the end tag is redundant and should be omitted ([this](https://developer.mozilla.org/en-US/docs/Web/HTML/Element/embed#technical_summary) is not the only reference that states this). #782 fixes this.
So it really looks like htmlproofer just got more picky about everything, also the reported error doesn't really help too much.

rhaschke, 2023-05-22:
Fixed via #782.

closed 2023-05-22 by rhaschke