moveit/moveit_tutorials #780: ### Description

christian-drax, 2023-05-20:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

christian-drax, 2023-05-20:
Having problems downloading all the packages

here is the error it gave me.
hristian@ubuntu:~/moveit_ws$ catkin build
--------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/christian/moveit_ws
--------------------------------------------------------------
Build Space:        [exists] /home/christian/moveit_ws/build
Devel Space:        [exists] /home/christian/moveit_ws/devel
Install Space:      [exists] /home/christian/moveit_ws/install
Log Space:          [exists] /home/christian/moveit_ws/logs
Source Space:       [exists] /home/christian/moveit_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        merged
--------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
--------------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------------
[build] Found 47 packages in 0.0 seconds.                                      
Starting  >>> geometric_shapes                                                 
Starting  >>> moveit_msgs                                                      
Starting  >>> moveit_resources_fanuc_description                               
Starting  >>> moveit_resources_panda_description                               
Starting  >>> moveit_resources_pr2_description                                 
Starting  >>> moveit_resources_prbt_support                                    
Starting  >>> rviz_visual_tools                                                
Starting  >>> srdfdom                                                          
Finished  <<< moveit_resources_pr2_description                               [ 0.3 seconds ]
Finished  <<< geometric_shapes                                               [ 0.3 seconds ]
Finished  <<< srdfdom                                                        [ 1.0 seconds ]
Finished  <<< moveit_resources_prbt_support                                  [ 0.6 seconds ]
Finished  <<< moveit_resources_fanuc_description                             [ 0.5 seconds ]
Starting  >>> moveit_resources_fanuc_moveit_config                             
Finished  <<< moveit_resources_panda_description                             [ 0.9 seconds ]
Starting  >>> moveit_resources_dual_panda_moveit_config                        
Starting  >>> moveit_resources_panda_moveit_config                             
Finished  <<< rviz_visual_tools                                              [ 3.8 seconds ]
Finished  <<< moveit_msgs                                                    [ 11.5 seconds ]
Finished  <<< moveit_resources_fanuc_moveit_config                           [ 0.2 seconds ]
Finished  <<< moveit_resources_dual_panda_moveit_config                      [ 0.2 seconds ]
Finished  <<< moveit_resources_panda_moveit_config                           [ 0.2 seconds ]
Starting  >>> moveit_resources                                                 
Starting  >>> moveit_core                                                      
Finished  <<< moveit_resources                                               [ 0.4 seconds ]
_______________________________________________________________________________
Errors     << moveit_core:make /home/christian/moveit_ws/logs/moveit_core/build.make.010.log
/home/christian/moveit_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp: In member function ‘moveit::core::JointModel* moveit::core::RobotModel::constructJointModel(const urdf::Joint*, const urdf::Link*, const srdf::Model&)’:
/home/christian/moveit_ws/src/moveit/moveit_core/robot_model/src/robot_model.cpp:999:71: error: ‘const class srdf::Model’ has no member named ‘getJointProperties’
  999 |     for (const auto& [property_name, property_value_str] : srdf_model.getJointProperties(joint_name))
      |                                                                       ^~~~~~~~~~~~~~~~~~
make[2]: *** [robot_model/CMakeFiles/moveit_robot_model.dir/build.make:180: robot_model/CMakeFiles/moveit_robot_model.dir/src/robot_model.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3655: robot_model/CMakeFiles/moveit_robot_model.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/christian/moveit_ws/build/moveit_core; catkin build --get-env moveit_core | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << moveit_core:make                                               [ Exited with code 2 ]
Failed    <<< moveit_core                                                    [ 5.2 seconds ]
[... 15 lines cut ...]
Abandoned <<< moveit_kinematics                                              [ Unrelated job failed ]
Abandoned <<< moveit_planners_ompl                                           [ Unrelated job failed ]
Abandoned <<< moveit_ros_move_group                                          [ Unrelated job failed ]
[... cut ...]
Abandoned <<< pilz_industrial_motion_planner                                 [ Unrelated job failed ]
[build] Summary: 12 of 47 packages succeeded.                                  
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 34 packages were abandoned.                               
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 17.5 seconds total.

rhaschke, 2023-05-20:
Next time, please enter the issue description in the first field (or edit it if you accidentally missed it) and also provide a title and the requested environment information.

Regarding your issue: You are trying to build MoveIt from source, which requires an updated version of srdfdom as well.
As you just want to build the tutorials, I guess, just remove the whole moveit source folder and use the moveit binaries instead.

closed 2023-05-21 by rhaschke