moveit/moveit_tutorials #783: Getting Started Tutorial: Catkin build fails with CMake Error

hedaniel7, 2023-05-24:
### Description

I followed the Getting Started Tutorial twice and produced errors when I tried `catkin build `. It failed twice with similar CMake Errors:

- Errors     << srdfdom:cmake /home/dan/ws_moveit/logs/srdfdom/build.cmake.000.log                                                                                                 
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR

- Errors     << rviz_visual_tools:cmake /home/dan/ws_moveit/logs/rviz_visual_tools/build.cmake.000.log (see console outputs posted later for details)                                                                             

 

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04.6 LTS (on Surface Book 3)

### Steps to reproduce
I followed this tutorial:
https://ros-planning.github.io/moveit_tutorials/doc/getting_started/getting_started.html

### Expected behaviour
It should not have any errors.

### Backtrace or Console outputs
1st attempt:
https://gist.github.com/hedaniel7/6c3e55b75568e4c193669beeb06ec849
2nd attempt:
https://gist.github.com/hedaniel7/b2cd9a24342abc0d90bd9d11e769d17c

rhaschke, 2023-05-24:
Looks like you are missing a correct boost installation. Please check that!

hedaniel7, 2023-05-25:
[`(base) dan@SurfaceBook3:~$ sudo apt-get install libboost-all-dev`](https://gist.github.com/hedaniel7/7d4b5369a8fd8835b50a1a45535da9f8)

The installation seemed fine. I even reinstalled it and I still get the same errors

rhaschke, 2023-05-25:
We cannot help more than pointing out possible issues. You need to check your boost installation. Verify that cmake finds the correct one. As you see from our [CI build](https://github.com/ros-planning/moveit_tutorials/actions/runs/5044927221), the code works as expected.