moveit/moveit_tutorials #786: Error: roslaunch urdf_tutorial display.launch model:=panda_multiple_arms.urdf

flydragon2018, 2023-06-08:
### Description

File "/opt/ros/noetic/lib/python3/dist-packages/joint_state_publisher_gui/__init__.py", line 126, in __init__
    self.center()
  File "/opt/ros/noetic/lib/python3/dist-packages/joint_state_publisher_gui/__init__.py", line 184, in center
    joint_info['slider'].setValue(self.valueToSlider(joint['zero'],joint))
TypeError: setValue(self, a0: int): argument 1 has unexpected type 'float'

Overview of your issue here.

### Your environment
* ROS Distro: noetic
* OS Version: Ubuntu 20.04
* Source or Binary build?  source
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce

follow the tutorial, until  roslaunch urdf_tutorial display.launch model:=panda_multiple_arms.urdf
https://ros-planning.github.io/moveit_tutorials/doc/multiple_robot_arms/multiple_robot_arms_tutorial.html

### Expected behaviour
none error message.

### Backtrace or Console output

File "/opt/ros/noetic/lib/python3/dist-packages/joint_state_publisher_gui/__init__.py", line 126, in __init__
    self.center()
  File "/opt/ros/noetic/lib/python3/dist-packages/joint_state_publisher_gui/__init__.py", line 184, in center
    joint_info['slider'].setValue(self.valueToSlider(joint['zero'],joint))
TypeError: setValue(self, a0: int): argument 1 has unexpected type 'float'

flydragon2018, 2023-06-08:
I am not sure if the logic is right.

In the related setValue(), 
I fixed it by  converting  the datatype to int, and  passing one parameter only.

so for ,   no error reported

rhaschke, 2023-06-13:
I cannot reproduce this issue. I just downloaded the `panda_multiple_arms.xacro` file and ran:
`roslaunch urdf_tutorial display.launch model:=panda_multiple_arms.xacro`

Also manually running xacro and loading the generated .urdf works as expected.
I'm closing this for now. If you still have issues, please feel free to reopen the issue and provide more details how to reproduce the problem.

closed 2023-06-13 by rhaschke