moveit/moveit_tutorials #792: How to use InformedRRTstar in Moveit?

lisacong, 2023-06-19:
### Description

I want to use InformedRRTstar in moveit. How should I configure in moveit？

### Your environment
* ROS Distro: melodic
* OS Version: e.g. Ubuntu 18.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

simonschmeisser, 2023-06-19:
From a quick glance it seems not to be integrated yet. That's not very hard however. Please have a look at this commit https://github.com/ros-planning/moveit/commit/51cff827c92709f3d0b2b6142e42c678dfbbd88a and try to do the same for InformedRRTstar. Then please open a pull request (PR) so everybody can enjoy it

lisacong, 2023-06-21:
I have achieved to call InforemedRRTstar in moveit. But now I don't know which optimization_objective to choice.

simonschmeisser, 2023-06-21:
Please provide a PR to add InformedRRTstar so others can profit as well. I don't understand your question unfortunately but generally you can add the optimization_objective to your ompl_planning.yaml as you can see here: https://ros-planning.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html#ompl-optimization-objectives

lisacong, 2023-06-25:
After few days research，I know that InformedRRTstar should choice "PathLengthOptimizationObjective" as optimization_objective in ompl_planning.yaml. InfomedRRTstar will use PathLengthDirectInfSampler as informed sample. But i encounter  a  problem: in the algorithm of PHS, the GoalSample should be used , but the goal state can not be sampled.