moveit/moveit_tutorials #793: Question on Planning with Custom Constraints (CHOMP)

leonardoedgar, 2023-07-02:
### Description

Hi, I want to run CHOMP as post-processing in my motion planning pipeline with custom constraints. Ideally, I will have a trajectory in the form `trajectory_msgs/JointTrajectory.msg` as the input to CHOMP and would like the output to be `trajectory_msgs/JointTrajectory.msg` as well. The custom constraint that I want to impose is the ZMP constraint that is represented in the form of:
$L(s) \leq a(s)\ddot{s} + b(s)\dot{s}^2 + c(s) \leq U(s)$
where $s$ is a path parameter $s \in [0, 1]$, $L$ and $U$ are the lower and upper bound of the constraint respectively, and $a$, $b$, $c$ are coefficients.

I think in the [CHOMP paper](https://www.ri.cmu.edu/pub_files/2009/5/icra09-chomp.pdf) the author also used ZMP constraints in a quadruped robot, so theoretically planning under ZMP constraints with CHOMP is possible.

Do you have any pointers on where I can implement this? Thanks

### Your environment
* ROS Distro: Kinetic/Melodic/Noetic
* OS Version: Ubuntu 16.04/18.04/20.04

simonschmeisser, 2023-07-02:
Please have a look here: https://ros-planning.github.io/moveit_tutorials/doc/planning_adapters/planning_adapters_tutorial.html

and for the actual code here: https://github.com/ros-planning/moveit/tree/noetic-devel/moveit_planners/chomp/chomp_optimizer_adapter

leonardoedgar, 2023-07-02:
Hi,
Thanks for the pointer and the quick reply. I see that the links show how to use CHOMP as a post-processing in a motion planning pipeline. However, do you have any extra pointers on how to input/implement custom constraints to CHOMP (e.g. ZMP constraints as I mentioned above)?

simonschmeisser, 2023-07-02:
You're welcome. I haven't looked closely at CHOMP yet so no, I can't help you any further unfortunately. But I think the code of the chomp plugin (the planner itself) is not that huge so you can most likely get an idea by just reading through it. Please write down anything you find strange, afaik there is room for improvement in the planner code. Don't hesitate to open issues or PRs. Good luck!