moveit/moveit_tutorials #795: Pick and Place Pipeline crash

Rodpc07, 2023-07-30:
### Description
Hi, I am using a ABB Yumi and a couple of weeks ago I was able to successfully execute the pick and place tutorial with this arm. Today I tried running the code again but it crashes as soon as it calls the pick() function from the move_group. The only thing I did since the day it works correctly was run the setup assistant and generate the package again but didn't change anything while inside the setup assistant. I leave bellow the backtrace and appreciate in advance to anyone who knows what might have happen. 

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Source 55d2019cd827c90724fec8e446bd8367bc4e3df6

### Code

```
#include <moveit/move_group_interface/move_group_interface.h>
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/planning_interface/planning_interface.h>

#include <moveit_visual_tools/moveit_visual_tools.h>

#include <visualization_msgs/Marker.h>
#include <interactive_markers/interactive_marker_server.h>

#include <tf2_geometry_msgs/tf2_geometry_msgs.h>

#include <iostream>
#include <string>

#include <memory>

using namespace std;

void pick(moveit::planning_interface::MoveGroupInterface &arm_mgi, moveit::planning_interface::MoveGroupInterface &gripper_mgi)
{
    // BEGIN_SUB_TUTORIAL pick1
    // Create a vector of grasps to be attempted, currently only creating single grasp.
    // This is essentially useful when using a grasp generator to generate and test multiple grasps.
    std::vector<moveit_msgs::Grasp> grasps;
    grasps.resize(1);

    // Actual Grasping Pose
    grasps[0].grasp_pose.header.frame_id = "yumi_base_link";
    grasps[0].grasp_pose.pose.position.x = 0.383890;
    grasps[0].grasp_pose.pose.position.y = -0.008552;
    grasps[0].grasp_pose.pose.position.z = 0.491650;
    grasps[0].grasp_pose.pose.orientation.x = 0.471129;
    grasps[0].grasp_pose.pose.orientation.y = 0.538489;
    grasps[0].grasp_pose.pose.orientation.z = 0.500098;
    grasps[0].grasp_pose.pose.orientation.w = 0.487822;

    // Pre Grasp (vector representing the direction of approach)
    grasps[0].pre_grasp_approach.direction.header.frame_id = "yumi_base_link";
    grasps[0].pre_grasp_approach.direction.vector.x = 1.0;
    grasps[0].pre_grasp_approach.min_distance = 0.095;
    grasps[0].pre_grasp_approach.desired_distance = 0.115;

    // Post Grasp (vector representing the direction of retreat)
    grasps[0].post_grasp_retreat.direction.header.frame_id = "yumi_base_link";
    grasps[0].post_grasp_retreat.direction.vector.z = 1.0;
    grasps[0].post_grasp_retreat.min_distance = 0.1;
    grasps[0].post_grasp_retreat.desired_distance = 0.25;

    // Open Gripper Pose
    for (const std::string joint_name : gripper_mgi.getVariableNames())
    {
        grasps[0].pre_grasp_posture.joint_names.push_back(joint_name);
    };

    grasps[0].pre_grasp_posture.points.resize(1);
    grasps[0].pre_grasp_posture.points[0].positions.resize(2);
    grasps[0].pre_grasp_posture.points[0].positions[0] = 0.025;
    grasps[0].pre_grasp_posture.points[0].positions[1] = 0.025;
    grasps[0].pre_grasp_posture.points[0].time_from_start = ros::Duration(0.5);
[... 225 lines cut ...]
#6  0x00007ffff46ad6b7 in collision_detection::CollisionEnvFCL::checkRobotCollisionHelper (this=0x55555632f6a0, req=..., res=..., 
    state=..., acm=0x7fffac00d550) at /home/rodpc/hri/src/moveit/moveit_core/collision_detection_fcl/src/collision_env_fcl.cpp:313
#7  0x00007ffff46ad1e9 in collision_detection::CollisionEnvFCL::checkRobotCollision (this=0x55555632f6a0, req=..., res=..., state=..., 
    acm=...) at /home/rodpc/hri/src/moveit/moveit_core/collision_detection_fcl/src/collision_env_fcl.cpp:290
#8  0x00007ffff5d1e887 in planning_scene::PlanningScene::checkCollision (this=0x55555566d620, req=..., res=..., robot_state=..., 
    acm=...) at /home/rodpc/hri/src/moveit/moveit_core/planning_scene/src/planning_scene.cpp:493
#9  0x00007fffc128d7df in planning_scene::PlanningScene::checkCollision (this=0x55555566d620, req=..., res=..., robot_state=..., 
    acm=...) at /home/rodpc/hri/src/moveit/moveit_core/planning_scene/include/moveit/planning_scene/planning_scene.h:490
#10 0x00007fffc128bd67 in pick_place::ReachableAndValidPoseFilter::isEndEffectorFree (this=0x7fffac0118c0, plan=
    std::shared_ptr<struct pick_place::ManipulationPlan> (use count 1, weak count 0) = {...}, token_state=...)
    at /home/rodpc/hri/src/moveit/moveit_ros/manipulation/pick_place/src/reachable_valid_pose_filter.cpp:103
#11 0x00007fffc128beee in pick_place::ReachableAndValidPoseFilter::evaluate (this=0x7fffac0118c0, 
    plan=std::shared_ptr<struct pick_place::ManipulationPlan> (use count 1, weak count 0) = {...})
    at /home/rodpc/hri/src/moveit/moveit_ros/manipulation/pick_place/src/reachable_valid_pose_filter.cpp:111
#12 0x00007fffc127f6b9 in pick_place::ManipulationPipeline::processingThread (this=0x7fffac009f70, index=0)
    at /home/rodpc/hri/src/moveit/moveit_ros/manipulation/pick_place/src/manipulation_pipeline.cpp:174
#13 0x00007fffc127ee6f in pick_place::ManipulationPipeline::<lambda()>::operator()(void) const (__closure=0x7fffac016c28)
    at /home/rodpc/hri/src/moveit/moveit_ros/manipulation/pick_place/src/manipulation_pipeline.cpp:117
#14 0x00007fffc1281130 in boost::detail::thread_data<pick_place::ManipulationPipeline::start()::<lambda()> >::run(void) (
    this=0x7fffac016af0) at /usr/include/boost/thread/detail/thread.hpp:120
#15 0x00007ffff574143b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
--Type <RET> for more, q to quit, c to continue without paging--c
#16 0x00007ffff7743609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#17 0x00007ffff7390133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

Rodpc07, 2023-07-31:
After some tinkering, I found that this crash is the result of building the workspace with **-DCMAKE_BUILD_TYPE** set to **DEBUG**, since it doesn't crash when set to **RELEASE**. I don't know if the release build is hiding an underlying problem that was only caught in debug.

rhaschke, 2023-07-31:
Your console output is very explicit about the underlying error:
`The given transform is not an isometry! Its linear part involves non-unit scaling. The scaling matrix diagonal differs from [1 1 1] by [ 1.3513e-06 1.18114e-06 1.28091e-06] but the required precision is 1e-12.
`
Do you feed any transforms with float precision only (instead of double)?

The reason that the crash only occurs in DEBUG mode, is that we [add some extra checks in debug mode](https://github.com/ros-planning/geometric_shapes/blob/1675760af4c6f98b2ff4e111b5f797ebfb394cb1/include/geometric_shapes/check_isometry.h#L109). So, that's expected.

rhaschke, 2023-07-31:
By the way, the recommended way to plan pick+place actions is to use MTC:
https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html

Rodpc07, 2023-07-31:
> Do you feed any transforms with float precision only (instead of double)?

Excuse my ignorance; to my knowledge, I am only running demo.launch from my yumi_moveit_config and the node from the code above, in which I don't believe I used any floats. That's why I am confused about what is causing this. If I run this in release, will it cause any problems?

> By the way, the recommended way to plan pick+place actions is to use MTC:
https://ros-planning.github.io/moveit_tutorials/doc/moveit_task_constructor/moveit_task_constructor_tutorial.html

I appreciate the advice and will be sure to use it in future applications, but for my project, I don't need anything much more complex than what is shown at the pick-and-place tutorial and don't have the time to study and tinker with how to implement the MTC for a dual-arm robot.

rhaschke, 2023-07-31:
> If I run this in release, will it cause any problems?

These tiny deviations will not cause any issues.

closed 2023-07-31 by Rodpc07