moveit/moveit_tutorials #802: Invalid <param> tag

ShroyerL, 2023-11-10:
### Description

While following the tutorial (moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.rst), running the following command in Step 1:

roslaunch panda_moveit_config demo.launch rviz_tutorial:=true

returns an "invalid <param> tag" error "Cannot load command parameter [robot_description]

### Your environment
* ROS Distro: 
                      Noetic
* OS Version: 
                      Windows 10
* Source or Binary build?
                      Binary 
                      https://moveit.ros.org/install/

rhaschke, 2023-11-10:
Sorry, but I can't reproduce your error. Using standard debian packages, the command launches without errors.

ShroyerL, 2023-11-10:
Is the issue that I am using windows? I thought that was the point of the moveit Windows download link I provided?

rhaschke, 2023-11-11:
I missed the Windows thing. Using chocolatey is outdated. I recommend using ROS in WSL:
https://github.com/rhaschke/lecture/wiki/WSL-install

ShroyerL, 2023-11-11:
I have that installed already. When I try to use install ROS commands from your link, it says:

"E: The repository 'http://packages.ros.org/ros/ubuntu jammy Release' does not have a Release file."
"E: Unable to locate package ros-noetic-desktop"
"E: Unable to locate package ros-noetic-agni-tf-tools"

Trying to follow the ROS install instructions from https://wiki.ros.org/noetic/Installation/Ubuntu also yields errors about release files not being available.

simonschmeisser, 2023-11-11:
You installed jammy/22.04 but ROS Noetic is only available for Ubuntu Focal/20.04

rhaschke, 2023-11-11:
Follow [these instructions](https://ros.packages.techfak.net) to install ROS One on Ubuntu 22.04.
Better, directly go for ROS 2 (if possible): https://docs.ros.org/en/rolling/Installation/Ubuntu-Install-Debians.html

ShroyerL, 2023-11-11:
ok, I installed Ubuntu 20.04.6 LTS and installed noetic. Everything looks good. In order to use Moveit, should I use noetic specific commands? https://moveit.ros.org/install/ talks about using melodic instead of noetic.

rhaschke, 2023-11-11:
When asked to install packages, just replace `melodic` or `noetic` with `one`, which are the names of ROS distros.
ROS one is based on noetic. Otherwise, everything should work as specified.

simonschmeisser, 2023-11-11:
@rhaschke 's comment would be right for 22.04/Jammy/ROS one.

Since you did the downgrade to 20.04, which I'd recommend for beginners, please simply replace any mention of melodic with noetic. However I did not see any mention of `melodic` on the link you gave unless I select `18.04`. Could you please be more specific where something is not up to date so we can fix it?

ShroyerL, 2023-11-11:
@simonschmeisser, I was originally trying to use the 'Windows 10' tab (https://moveit.ros.org/install/) before I found out to use Ubuntu. It says melodic there.

when trying to use the following:
git clone https://github.com/ros-planning/moveit_tutorials.git -b master
git clone https://github.com/ros-planning/panda_moveit_config.git -b noetic-devel

It says:
"chmod on .git/config.lock failed: Operation not permitted"
"fatal: could not set 'core.filemode' to 'false'"

simonschmeisser, 2023-11-11:
[image]

all Noetic here for me

ShroyerL, 2023-11-11:
@simonschmeisser, click the Windows 10 button at the top right of your pic.

ShroyerL, 2023-11-11:
I got to the point of trying to launch moveit with:

roslaunch panda_moveit_config demo.launch rviz_tutorial:=true

It looks like the terminal is doing everything correctly, but no screen opens. Am I missing something?

[image]

ShroyerL, 2023-11-11:
there's something wrong with rviz

[image]

closed 2023-11-17 by ShroyerL