moveit/moveit_tutorials #803: Exit Code -11 when running Rviz

ShroyerL, 2023-11-17:
### Description

While trying to launch the panda tutorial using WSL2,
(https://ros-planning.github.io/moveit_tutorials/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html)

Rviz opens for a second before crashing with the following error:
[image]

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04.6 LTS
* Source or Binary build?
Binary

rhaschke, 2023-11-18:
Can you run other OpenGL applications, e.g. `glxinfo` or `glxgears`?

ShroyerL, 2023-11-18:
I can run glxinfo but not glxgears. glxgears returns a segmentation fault.

rhaschke, 2023-11-18:
Then, please approach the WSL2 developers first. There seems to be a more fundamental issue with your graphics hardware.