moveit/moveit_tutorials #805: Panda hand missing

CleisonManriqueAguirre, 2023-12-06:
### Description

There is no group named %s" % group

### Your environment
* ROS Distro: [Melodic]
* OS Version: e.g. Ubuntu 18.04

### Steps to reproduce
While running the tutorial.= and this command : rosrun moveit_tutorials move_group_python_interface_tutorial.py . I even changed to panda_hand and nothing happens

### Expected behaviour
The robot should take the box and then detach

### Backtrace or Console output

Traceback (most recent call last):
  File "/home/cleison/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py", line 510, in <module>
    main()
  File "/home/cleison/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py", line 488, in main
    tutorial.attach_box()
  File "/home/cleison/ws_moveit/src/moveit_tutorials/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py", line 401, in attach_box
    touch_links = robot.get_link_names(group=grasping_group)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/robot.py", line 234, in get_link_names
    raise MoveItCommanderException("There is no group named %s" % group)
moveit_commander.exception.MoveItCommanderException: There is no group named hand

CleisonManriqueAguirre, 2023-12-06:
solved: changed name from hand to panda_hands

closed 2023-12-06 by CleisonManriqueAguirre

law1209, 2024-11-14:
> solved: changed name from hand to panda_hands

panda_hand

rhaschke, 2024-11-14:
The name of the hand group was fixed already in Oct 2022:
https://github.com/moveit/moveit_tutorials/blob/426af1eb8adcbd5d77d1d822dd92a2b48b3f7a08/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L424