moveit/moveit_tutorials #806: Errors When adding EIT* to moveit

yw1055, 2023-12-11:
Hello,
  I am trying to add EIT* to move it!, and integrate our planner later, but I encountered the following error:
 
Starting  >>> pilz_industrial_motion_planner                                                                                                                                                            
________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << moveit_planners_ompl:make /home/yiwang/logs/moveit_planners_ompl/build.make.025.log                                                                                                       
/home/yiwang/src/moveit/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp:74:10: fatal error: ompl/geometric/planners/informedtrees/EITstar.h: No such file or directory
   74 | #include <ompl/geometric/planners/informedtrees/EITstar.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ompl_interface/CMakeFiles/moveit_ompl_interface.dir/build.make:90: ompl_interface/CMakeFiles/moveit_ompl_interface.dir/src/planning_context_manager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2545: ompl_interface/CMakeFiles/moveit_ompl_interface.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /home/yiwang/build/moveit_planners_ompl; catkin build --get-env moveit_planners_ompl | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

........................................................................................................................................................................................................
Failed     << moveit_planners_ompl:make                                      [ Exited with code 2 ]                                                                                                     
Failed    <<< moveit_planners_ompl                                           [ 2.6 seconds ]                                                                                                            
Abandoned <<< moveit                                                         [ Unrelated job failed ]                                                                                                   
Abandoned <<< moveit_planners                                                [ Unrelated job failed ]                                                                                                   
Abandoned <<< moveit_ros                                                     [ Unrelated job failed ]                                                                                                   
Abandoned <<< moveit_runtime                                                 [ Unrelated job failed ]                                                                                                   
Abandoned <<< panda_moveit_config                                            [ Unrelated job failed ]                                                                                                   
Abandoned <<< moveit_tutorials                                               [ Unrelated job failed ]                                                                                                   
Finished  <<< moveit_servo                                                   [ 0.8 seconds ]                                                                                                            
Finished  <<< pilz_industrial_motion_planner                                 [ 0.4 seconds ]                                                                                                            
Finished  <<< moveit_setup_assistant                                         [ 0.4 seconds ]                                                                                                            
Finished  <<< moveit_ros_benchmarks                                          [ 17.4 seconds ]                                                                                                           
[build] Summary: 41 of 48 packages succeeded.                                                                                                                                                           
[build]   Ignored:   None.                                                                                                                                                                              
[build]   Warnings:  None.                                                                                                                                                                              
[build]   Abandoned: 6 packages were abandoned.                                                                                                                                                         
[build]   Failed:    1 packages failed.                                                                                                                                                                 
[build] Runtime: 22.6 seconds total. 

Has anyone else faced this issue? If so, could you share how you resolved it?
Any help or suggestions would be greatly appreciated!
Thank you!

rhaschke, 2023-12-11:
Obviously you are using a too old version of OMPL:
Missing include: `ompl/geometric/planners/informedtrees/EITstar.h`

rhaschke, 2023-12-11:
OMPL 1.6 (the default in Focal/Noetic) doesn't include EITstar.h:
```
> ls /opt/ros/noetic/include/ompl-1.6/ompl/geometric/planners/informedtrees
ABITstar.h  aitstar  AITstar.h  bitstar  BITstar.h
```
Check that the file actually exists and that you MoveIt is using the OMPL version you are thinking it should use.

yw1055, 2023-12-11:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> CAUTION: This email originated from outside of the University System. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I cloned the ompl into user/src:
 ls /home/user/src/ompl/src/ompl/geometric/planners/informedtrees
ABITstar.h  AITstar.h  BITstar.h   eitstar    src
aitstar     bitstar    EIRMstar.h  EITstar.h

Do you mean I have to link to /opt/ros/noetic/include/ ?
Thanks a lot for your help!

rhaschke, 2023-12-11:
You need to point the MoveIt build to your own OMPL build by passing -DOMPL_DIR=... to cmake.

yw1055, 2023-12-11:
ok, thanks a lot for your suggestion.

simonschmeisser, 2023-12-11:
Simply putting everything in a catkin workspace and cleaning and recompiling everything should also work

yw1055, 2023-12-11:
> […](#)
> -- Gesendet von meinem Sailfish

I am trying to follow your suggestion, but when I clean build and develop, but got new errors:
/home/user/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h:39:10: fatal error: ompl/base/StateSpace.h: No such file or directory
   39 | #include <ompl/base/StateSpace.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
But, thanks your suggestion and I have to fix these errors.

yw1055, 2023-12-11:
> You need to point the MoveIt build to your own OMPL build by passing -DOMPL_DIR=... to cmake.

> I think I am using the latest version！😄 Get Outlook for iOS<https://aka.ms/o0ukef>
> […](#)

Hi Robert,

I appreciate your advice on integrating OMPL with MoveIt. I tried executing catkin config -DOMPL_DIR=/home/user/src/ompl as you suggested, but I encountered issues. I'm familiar with OMPL but new to MoveIt, and I'm finding some aspects of the setup challenging. Could you possibly provide more detailed instructions or clarify the steps needed to correctly configure the OMPL path in MoveIt? Any additional guidance you can offer would be immensely helpful to me.

Thank you so much

yw1055, 2023-12-12:
> You need to point the MoveIt build to your own OMPL build by passing -DOMPL_DIR=... to cmake.
Hi Robert,
Thanks for your response. I tried following your suggestion about configuring OMPL with MoveIt using "cmake -DOMPL_DIR=/home/user/src/ompl " , but I'm still facing challenges. I'm quite familiar with OMPL but new to MoveIt. Would you be able to provide more detailed instructions or steps? Additionally, I've been trying to locate where the path is set to /opt/ros/noetic/include/, but haven't had success with that either. Any further guidance would be greatly appreciated.
Thank you so much for your help!

rhaschke, 2023-12-12:
This is nothing specific to MoveIt. You just need to apply standard cmake rules. Thus, if you _installed_ OMPL, you need to point OMPL_DIR to the location of the installed `omplConfig.cmake` file.
Simon pointed out an even simpler solution: putting both OMPL and MoveIt into your catkin workspace and building everything from there.

yw1055, 2023-12-12:
> This is nothing specific to MoveIt. You just need to apply standard cmake rules. Thus, if you _installed_ OMPL, you need to point OMPL_DIR to the location of the installed `omplConfig.cmake` file. [...]

I appreciate your suggestion and have attempted to follow it, but unfortunately, the issue persists. It appears that integrating a custom planner into MoveIt is more complex than implementing a planner directly in OMPL. Any further guidance or assistance would be greatly appreciated.

simonschmeisser, 2023-12-12:
It should be as simple as
```
mkdir -p ws/src
cd ws/src
git clone https://github.com/ompl/ompl
git clone https://github.com/ros-planning/moveit
source /opt/ros/noetic/setup.bash
cd ..
catkin build
```
if you still run into issues please provide more details like `catkin config`, logs, etc.

yw1055, 2023-12-12:
> It should be as simple as
> 
> ```
> mkdir -p ws/src
> cd ws/src
> git clone https://github.com/ompl/ompl
> git clone https://github.com/ros-planning/moveit
> source /opt/ros/noetic/setup.bash
> cd ..
> catkin build
> ```
> 
> if you still run into issues please provide more details like `catkin config`, logs, etc.

Errors     << moveit_planners_ompl:make /home/yiwang/logs/moveit_planners_ompl/build.make.073.log
In file included from /home/yw1055/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/model_based_planning_context.h:39,
                 from /home/yw1055/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/planning_context_manager.h:39,
                 from /home/yw1055/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/ompl_interface.h:39,
                 from /home/yiwang/src/moveit/moveit_planners/ompl/ompl_interface/src/ompl_interface.cpp:37:
/home/yw1055/src/moveit/moveit_planners/ompl/ompl_interface/include/moveit/ompl_interface/parameterization/model_based_state_space.h:39:10: fatal error: ompl/base/StateSpace.h: No such file or directory
   39 | #include <ompl/base/StateSpace.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ompl_interface/CMakeFiles/moveit_ompl_interface.dir/build.make:76: ompl_interface/CMakeFiles/moveit_ompl_interface.dir/src/ompl_interface.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2545: ompl_interface/CMakeFiles/moveit_ompl_interface.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
cd /home/yw1055/build/moveit_planners_ompl; catkin build --get-env moveit_planners_ompl | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << moveit_planners_ompl:make                                      [ Exited with code 2 ]
Failed    <<< moveit_planners_ompl                                           [ 3.3 seconds ]
Abandoned <<< moveit                                                         [ Unrelated job failed ]
Abandoned <<< moveit_planners                                                [ Unrelated job failed ]
Abandoned <<< moveit_plugins                                                 [ Unrelated job failed ]
Abandoned <<< moveit_commander                                               [ Unrelated job failed ]
Abandoned <<< moveit_ros                                                     [ Unrelated job failed ]
Abandoned <<< moveit_runtime                                                 [ Unrelated job failed ]
Abandoned <<< panda_moveit_config                                            [ Unrelated job failed ]
Abandoned <<< moveit_ros_move_group                                          [ Unrelated job failed ]
Abandoned <<< moveit_ros_manipulation                                        [ Unrelated job failed ]
Abandoned <<< moveit_ros_benchmarks                                          [ Unrelated job failed ]
Abandoned <<< moveit_ros_planning_interface                                  [ Unrelated job failed ]
Abandoned <<< moveit_planners_chomp                                          [ Unrelated job failed ]
Abandoned <<< moveit_ros_visualization                                       [ Unrelated job failed ]
Abandoned <<< moveit_servo                                                   [ Unrelated job failed ]
Abandoned <<< moveit_setup_assistant                                         [ Unrelated job failed ]
Abandoned <<< moveit_tutorials                                               [ Unrelated job failed ]
Abandoned <<< pilz_industrial_motion_planner_testutils                       [ Unrelated job failed ]
Abandoned <<< pilz_industrial_motion_planner                                 [ Unrelated job failed ]
Finished  <<< moveit_fake_controller_manager                                 [ 20.1 seconds ]                                                                  
Finished  <<< moveit_ros_robot_interaction                                   [ 37.7 seconds ]                                                                  
Finished  <<< moveit_ros_perception                                          [ 1 minute and 3.1 seconds ]                                                      
Finished  <<< moveit_visual_tools                                            [ 1 minute and 10.0 seconds ]                                                     
Finished  <<< moveit_kinematics                                              [ 1 minute and 11.3 seconds ]                                                     
Finished  <<< moveit_ros_warehouse                                           [ 1 minute and 11.5 seconds ]                                                     
[build] Summary: 30 of 49 packages succeeded.                                                                                                                  
[build]   Ignored:   None.                                                                                                                                     
[build]   Warnings:  30 packages succeeded with warnings.                                                                                                      
[build]   Abandoned: 18 packages were abandoned.                                                                                                               
[... cut ...]
Thank you for your suggestion. I've been following the steps you outlined, but I'm still encountering errors similar to those I experienced yesterday. Specifically, I attempted to point OMPL_INCLUDE_DIRS in src/moveit/moveit_planners/ompl/CMakeLists.txt to /usr/local/include/ompl-1.6/ompl this morning. I made this change because I noticed that EITstar.h is located in that directory. Despite this adjustment, the errors persist.

Could you provide any further guidance or suggestions? Resolving this issue is crucial for me, as the manipulator problem is the last experimental comparison I need to make with the current state-of-the-art planners.

Thank you so much for your help.

simonschmeisser, 2023-12-13:
I assumed you have all MoveIt packages and dependencies installed as debian packages as well.

Did you clone and build ompl as well? Please provide the full output via https://gist.github.com/ or similar

yw1055, 2023-12-13:
> I assumed you have all MoveIt packages and dependencies installed as debian packages as well.
> Yes, I have installed all MoveIt packages and their dependencies as debian packages. I found that merely following your seven-line suggestion led to errors, such as the absence of 'geometric_shapes'. 
> Did you clone and build ompl as well? Please provide the full output via https://gist.github.com/ or similar
> I have only cloned OMPL but have not proceeded with building it. I followed your instructions precisely, since I am concerned about potential errors if I deviate from these instructions.

yw1055, 2023-12-13:
> I assumed you have all MoveIt packages and dependencies installed as debian packages as well.
> Yes, I have installed all MoveIt packages and their dependencies as debian packages. I found that merely [...]

Thanks a lot for your time and patience.

yw1055, 2023-12-19:
For anyone who will meet the same issue:
  I already successfully added EITstar to moveit! Following is my resolution:
 1. mkdir -p build/Release
 2. cd build/Release
 3. cmake ../..
 4. cmake -DCMAKE_INSTALL_PREFIX=/opt/ros/noetic ../..
 5. go to ~/src/moveit/moveit_planners/ompl. In CmakeLists.txt, change find_package(ompl REQUIRED) to change find_package(ompl)
 6. cd ~/src/moveit/moveit_planners/ompl/ompl_interface, after line1 add set(OMPL_LIBRARIES "/opt/ros/noetic/lib/x86_64-linux-gnu/libompl.so")
 7. following above should work

yw1055, 2023-12-19:
> It should be as simple as
> 
> ```
> mkdir -p ws/src
> cd ws/src
> git clone https://github.com/ompl/ompl
> git clone https://github.com/ros-planning/moveit
> source /opt/ros/noetic/setup.bash
> cd ..
> catkin build
> ```
> 
> if you still run into issues please provide more details like `catkin config`, logs, etc.

I have solved this problem and post my resolution, but I got another issues, do you have any suggestions:
when I run EIT* on Moveit, it shows the following errors:

[ INFO] [1702963608.130321083]: Using planning pipeline 'ompl'
[ERROR] [1702963608.137822379]: Unknown planner: 'geometric::EITstar'
[ INFO] [1702963608.137979514]: Planner configuration 'panda_arm[EITstar]' will use planner 'geometric::EITstar'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [1702963608.138928178]: Exception caught executing adapter 'Fix Start State Path Constraints': bad_function_call
Skipping adapter instead.
[ WARN] [1702963608.146283620]: Goal sampling thread never did any work.
[ERROR] [1702963608.149505999]: Unknown planner: 'geometric::EITstar'
[ INFO] [1702963608.149638935]: Planner configuration 'panda_arm[EITstar]' will use planner 'geometric::EITstar'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [1702963608.150060330]: Exception caught executing adapter 'Fix Start State In Collision': bad_function_call
Skipping adapter instead.
[ WARN] [1702963608.158429168]: Goal sampling thread never did any work.
[ERROR] [1702963608.162115396]: Unknown planner: 'geometric::EITstar'
[ INFO] [1702963608.162285068]: Planner configuration 'panda_arm[EITstar]' will use planner 'geometric::EITstar'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [1702963608.162727207]: Exception caught executing adapter 'Fix Start State Path Constraints': bad_function_call
Skipping adapter instead.
[ WARN] [1702963608.170903737]: Goal sampling thread never did any work.
[ERROR] [1702963608.174163551]: Unknown planner: 'geometric::EITstar'
[ INFO] [1702963608.174270955]: Planner configuration 'panda_arm[EITstar]' will use planner 'geometric::EITstar'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [1702963608.174698925]: Exception caught executing adapter 'Fix Start State Bounds': bad_function_call
Skipping adapter instead.
[ WARN] [1702963608.182978372]: Goal sampling thread never did any work.
[ERROR] [1702963608.186507855]: Unknown planner: 'geometric::EITstar'
[ INFO] [1702963608.186623227]: Planner configuration 'panda_arm[EITstar]' will use planner 'geometric::EITstar'. Additional configuration parameters will be set when the planner is constructed.
[ERROR] [1702963608.187023258]: Exception caught executing ad

yw1055, 2023-12-19:
> This is nothing specific to MoveIt. You just need to apply standard cmake rules. Thus, if you _installed_ OMPL, you need to point OMPL_DIR to the location of the installed `omplConfig.cmake` file. [...]

Iusse has been solved. Thanks for your previous suggestion.

rhaschke, 2023-12-19:
> `[ERROR] [1702963608.137822379]: Unknown planner: 'geometric::EITstar'`

You need to register the planner with MoveIt:
https://github.com/ros-planning/moveit/tree/master/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L304-L306

simonschmeisser, 2023-12-19:
Your solution looks very hacky to me, the official why like I wrote it or how it is written in the documentation should work. But without logs it is hard to help you

yw1055, 2023-12-19:
> > `[ERROR] [1702963608.137822379]: Unknown planner: 'geometric::EITstar'`
> 
> You need to register the planner with MoveIt: https://github.com/ros-planning/moveit/tree/master/moveit_planners/ompl/ompl_interface/src/planning_context_manager.cpp#L304-L306

The errors occur after registering the planner.

yw1055, 2023-12-19:
> Your solution looks very hacky to me, the official why like I wrote it or how it is written in the documentation should work. But without logs it is hard to help you

Thank you for your input. I understand that the method you suggested aligns with the official documentation, and under normal circumstances, it should work effectively. However, when I tried implementing it, I ran into the above errors.

yw1055, 2023-12-19:
> For anyone who will meet the same issue: I already successfully added EITstar to moveit! Following is my resolution:
> 
> 1. mkdir -p build/Release
> 2. cd build/Release
> 3. cmake ../..
> 4. cmake -DCMAKE_INSTALL_PREFIX=/opt/ros/noetic ../..
> 5. go to ~/src/moveit/moveit_planners/ompl. In CmakeLists.txt, change find_package(ompl REQUIRED) to change find_package(ompl)
> 6. cd ~/src/moveit/moveit_planners/ompl/ompl_interface, after line1 add set(OMPL_LIBRARIES "/opt/ros/noetic/lib/x86_64-linux-gnu/libompl.so")
> 7. following above should work
>    Mine works. I hope this resolution can help and save your time.
> 
> **EDIT from maintainer: The suggested modifications to MoveIt cmake files are not needed! If so, cmake wasn't able to find OMPL correctly.**

yw1055, 2024-01-22:
A new update:
Hi Robert and Simon,
I have been working on another project and already submitted my results to IJCAI, so now I'm back to working on the sampling-based planner. I reinstalled OMPL and MoveIt, and it looks like Simon's previous suggestion kind of works, but only when using catkin_make_isolated. Anyway, I finally managed to add a custom planner (EIT*) to MoveIt 20 minutes ago. The new problem is a symbol lookup error in move_group: 

[ INFO] [1705899674.055363957]: Using planning pipeline 'ompl'
/home/yiwang/new_ws/devel_isolated/moveit_ros_move_group/lib/moveit_ros_move_group/move_group: symbol lookup error: /home/yiwang/new_ws/devel_isolated/moveit_planners_ompl/lib/libmoveit_ompl_interface.so.1.1.13: undefined symbol: _ZN4ompl5tools9BenchmarkC1ERNS_9geometric11SimpleSetupERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
[move_group-5] process has died [pid 54234, exit code 127, cmd /home/yiwang/new_ws/devel_isolated/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/yiwang/.ros/log/2576f3de-b8e3-11ee-9347-fbcb5429de13/move_group-5.log].
log file: /home/yiwang/.ros/log/2576f3de-b8e3-11ee-9347-fbcb5429de13/move_group-5*.log

I have less experience with MoveIt, so do you have any suggestions on how to resolve this error?
Thank you.

closed 2024-01-23 by yw1055