moveit/moveit_tutorials #807: MoveIt Setup Assistant tutorial: Spawn service timed out error

marinaaoki, 2023-12-22:
### Description

I have been trying to spawn the robot in Gazebo using the URDF generated by the MoveIt Setup Assistant using step 9 in the [tutorial](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html). 

Launching the empty world works without any issues, but spawning the model in the next step simply gives me an error that the spawn service timed out.

I have followed the steps in the tutorial as specified.

### Your environment
* ROS Distro: Noetic
* OS Version: Ubuntu 20.04
* Binary build 
* 1.1.13-2focal.20231204.184529

### Steps to reproduce
By following the MoveIt Setup Assistant tutorial steps 1-9

### Expected behaviour
Robot model is spawned in Gazebo without any error

### Backtrace or Console output

Spawning the model using the command `rosrun gazebo_ros spawn_model -file </path_to_new_urdf/file_name.urdf> -urdf -x 0 -y 0 -z 1 -model panda` gives me the following error message:

[INFO] [1703238121.019351]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name panda
[ERROR] [1703238121.020110]: Spawn service failed. Exiting.

In the other terminal where the empty world was launched I get this error message:

Error [parser.cc:488] parse as old deprecated model file failed.
Error Code 4 Msg: Required attribute[filename] in element[plugin] is not specified in SDF.
Error Code 8 Msg: Error reading element <plugin>
Error Code 8 Msg: Error reading element <model>
Error Code 8 Msg: Error reading element <sdf>

ReemAlhosainy, 2024-02-24:
Have you found a solution to this problem as I face the same?

rhaschke, 2024-02-25:
The `gazebo/plugin` tag added by MSA misses the `filename="libgazebo_ros_control.so"` attribute.
This is fixed via https://github.com/ros-planning/moveit/pull/3572.

closed 2024-02-25 by rhaschke

erwinmaast, 2024-02-27:
> The `gazebo/plugin` tag added by MSA misses the `filename="libgazebo_ros_control.so"` attribute. This is fixed via [ros-planning/moveit#3572](https://github.com/ros-planning/moveit/pull/3572).

After Adding this missing filename, the robot spawns, but falls (ragdolls) on the ground.

How to fix this ?

rhaschke, 2024-02-27:
Do you have ROS controllers activated? Which robot you use? The Panda robot stays upright as expected.

erwinmaast, 2024-02-27:
I followed the same steps as described above. I used the panda.urdf.xacro from the franka_description package. After generating the panda_movit_config package trough the assistant, I try to launch Rviz and Gazebo. 

`roslaunch panda_moveit_config demo_gazebo.launch` 

Before adding the missing filename the robot didn't spawn. (Parsing Error as described above)
After adding the filename and running the roslaunch command the robot ragdolls on the ground. The base link (panda_link0) isn't even connected with the ground. 

I checked everything in Setup Assistant a dozen times already. I tried different settings, but still same outcome.

rhaschke, 2024-02-27:
Running `roslaunch panda_moveit_config demo_gazebo.launch` works out-of-the-box with the existing panda_moveit_config. There is no need to modify any files with MSA. If you made changes to the package, revert them and try again.