moveit/moveit_tutorials #815: Pick-Place-tutorial objects not visible in Gazebo

Kamlesh364, 2024-03-01:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Melodic
* OS Version: e.g. Ubuntu 18.04

### Steps to reproduce
`roslaunch panda_moveit_config demo_gazebo.launch`
`rosrun moveit_tutorials pick_place_tutorial`

### Expected behaviour
All the motion planning and execution being done successfully. The only issue is that after running `pick_place_tutorial` node, the object, table0, and table1 are visible in Rviz and not in Gazebo.

@moriarty @dbc @kylc @tnaka @mintar

rhaschke, 2024-03-01:
This is by design. Collision objects are known by MoveIt only and visualized in rviz by MoveIt's plugins. Gazebo doesn't know about these.

closed 2024-03-01 by rhaschke

Kamlesh364, 2024-03-01:
could you please guide me, how to add them to Gazebo simultaneously with RVIZ?

@rhaschke

rhaschke, 2024-03-02:
https://wiki.ros.org/simulator_gazebo/Tutorials/SpawningObjectInSimulation