moveit/moveit_tutorials #820: object with no geometry

moccij, 2024-04-30:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: [Indigo|Jade|Kinetic]
* OS Version: e.g. Ubuntu 16.04
* Source or Binary build?
* If binary, which release version?
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2024-04-30:
There is no description for this issue! Please edit.

moccij, 2024-04-30:
Sorry, I did not intend to submit an issue, this can be removed

closed 2024-04-30 by moccij