moveit/moveit_tutorials #821: github404

ShanningZhuang, 2024-05-05:
The document of moveit 1 is 404.
[ros-planning.github.io/moveit_tutorials/](https://ros-planning.github.io/moveit_tutorials/)

rhaschke, 2024-05-05:
The tutorials were migrated to https://moveit.github.io/moveit_tutorials. Where did you find the broken link?

ShanningZhuang, 2024-05-06:
> The tutorials were migrated to https://moveit.github.io/moveit_tutorials. Where did you find the broken link?

Thanks! The link you gave me is correct. I found the broken link in this repository's about and in google.

closed 2024-05-06 by ShanningZhuang

torydebra, 2024-07-30:
The old link is also present on the moveit2 page https://moveit.picknik.ai/main/index.html, at the top there is `see [MoveIt 1 tutorials](https://ros-planning.github.io/moveit_tutorials).`