moveit/moveit_tutorials #824: the moveit tutorial is depracted?

NANDKLFSAJDKJHAjkhdsakxz1312, 2024-06-08:
### Description

Overview of your issue here.
this teaches how to integrate with gazebo, but the content has changed in the repo mentioned here, namely, the franka_ros and panda_moveit_config repo, I can not find the right way to modify the files.
### Your environment
* ROS Distro: noetic
* OS Version: e.g. Ubuntu 20.04
* Source or Binary build? binary
* If binary, which release version? 
* If source, which git commit or tag?

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

rhaschke, 2024-06-09:
Unfortunately, your description doesn't provide detailed enough information to understand your issue. Please explain, which tutorial you are referring to and which information is missing / has changed.

NANDKLFSAJDKJHAjkhdsakxz1312, 2024-06-09:
Sorry for that. I refer to the [tutorial](https://moveit.github.io/moveit_tutorials/doc/gazebo_simulation/gazebo_simulation.html).
From this tutorial I try to use the ready-made work from the superlink it provides, namely, at the very last pharagraph, but they both do not even have demo_gazebo.launch file, so I am wondering whtrther this tutorial is out of date or not.

rhaschke, 2024-06-14:
The tutorial is heavily outdated. If you just want to start MoveIt+Gazebo for the Panda robot, just launch:
`roslaunch panda_moveit_config demo_gazebo.launch`
using the default binary packages.
I modified the tutorial to be more generic.

closed 2024-06-14 by rhaschke