moveit/moveit_tutorials #827: Compiling the noetic version according to the official docs reports errors

lblbk, 2024-09-09:
### Description

according the official docs [download-example-code](https://moveit.github.io/moveit_tutorials/doc/getting_started/getting_started.html#download-example-code)

### Your environment
* ROS Distro: [noetic]
* OS Version: Ubuntu 20.04
* Source build
* If binary, which release version?  `git clone https://github.com/moveit/moveit_tutorials.git -b master`

### Steps to reproduce
nothing todo, just according the docs well reports errors

### Backtrace or Console output
```
move_group_interface_tutorial.cpp: In function ‘int main(int, char**)’:
/home/wibot/Work/robot/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp:305:94: error: no matching function for call to ‘moveit::planning_interface::MoveGroupInterface::computeCartesianPath(std::vector<geometry_msgs::Pose_<std::allocator<void> > >&, const double&, moveit_msgs::RobotTrajectory&)’
  305 |   double fraction = move_group_interface.computeCartesianPath(waypoints, eef_step, trajectory);
      |                                                                                              ^
In file included from /home/wibot/Work/robot/ws_moveit/src/moveit_tutorials/doc/move_group_interface/src/move_group_interface_tutorial.cpp:37:
/opt/ros/noetic/include/moveit/move_group_interface/move_group_interface.h:782:10: note: candidate: ‘double moveit::planning_interface::MoveGroupInterface::computeCartesianPath(const std::vector<geometry_msgs::Pose_<std::allocator<void> > >&, double, double, moveit_msgs::RobotTrajectory&, bool, moveit_msgs::MoveItErrorCodes*)’
  782 |   double computeCartesianPath(const std::vector<geometry_msgs::Pose>& waypoints, double eef_step, double jump_threshold,
      |          ^~~~~~~~~~~~~~~~~~~~
/opt/ros/noetic/include/moveit/move_group_interface/move_group_interface.h:782:10: note:   candidate expects 6 arguments, 3 provided
```

### solution
function `computeCartesianPath` have 4 arguments in the moledic-devel branch [L281](https://github.com/moveit/moveit_tutorials/blob/f77a29c7c3281ae82c6a4d793c39bed61c45cfb4/doc/move_group_interface/src/move_group_interface_tutorial.cpp#L281), but only 3 arguments in the default branch

```
  const double jump_threshold = 0.0;
  const double eef_step = 0.01;
  double fraction = move_group_interface.computeCartesianPath(waypoints, eef_step, jump_threshold, trajectory);
```

**Just copy the code from the melodic version**

rhaschke, 2024-09-09:
If you are building against MoveIt source, you will need the 3-arguments version. Probably you are building against the binary release.

lblbk, 2024-09-09:
> If you are building against MoveIt source, you will need the 3-arguments version. Probably you are building against the binary release.

I am following the tutorial step by step. [getting_started.html](https://moveit.github.io/moveit_tutorials/doc/getting_started/getting_started.html)

rhaschke, 2024-09-09:
The tutorial suggests building MoveIt from source. This should work w/o errors.
Your compiler picks up MoveIt sources from /opt/ros/noetic instead.

closed 2024-09-19 by lblbk

CAN-Lee, 2025-02-03:
Also encounter this problem. The solution works.