moveit/moveit_tutorials #834: How to make `compute_cartesian_path` to not go to base pose after reaching a waypoint

yaswanth1701, 2024-11-14:
Hi everyone, I am new to moveit and I am using `compute_carstesian_path` to generate joint trajectory to reach a set of Cartesian waypoints. I observed from using it in both movieit franka tutorials and on real robot is the manipulator always goes  to the base ee pose (zero joints position)  before going to the next waypoint which I don't want to happen. I just want my end effector to reach next waypoint from the currently achieved/reached waypoint instead of going to the base pose and then to the next waypoint. Also, I tried to reach waypoints using `move_group.go(pose)`  but I am not able to achieve my desired behaviour on it due to some hardware issue.

rhaschke, 2024-11-14:
This is a duplicate of https://github.com/moveit/moveit/issues/3651. Please upgrade to MoveIt 1.1.16.

yaswanth1701, 2024-11-14:
Thanks a lot for the help this solves my issue

yaswanth1701, 2024-11-14:
One more issue which I noticed just now is moveit generates partial trajectory sometimes and sometimes full trajectories. I am not sure to why is this happening, is there any limit on number of waypoints which can passed to `compute_cartesian_path` ?

rhaschke, 2024-11-14:
`compute_cartesian_path` is not a planner (avoiding obstacles) but a _motion generator_: it just generates a straight-line motion in Cartesian space. If that's not possible (due to joint limits or obstacles), it just returns the first part of the trajectory, which was feasible.
OMPL is definitely not involved for `compute_cartesian_path`! If you want to plan a trajectory with OMPL, you need to use the planning capabilities of move_group, which only handle a single segment. For multiple waypoints, checkout [MTC](https://github.com/moveit/moveit_task_constructor) or alternative [Cartesian planners](https://picknik.ai/cartesian%20planners/moveit/motion%20planning/2021/01/07/guide-to-cartesian-planners-in-moveit.html).

yaswanth1701, 2024-11-14:
Got it!! thanks a lot for the help I will look into the provided resources.

closed 2024-11-14 by yaswanth1701