moveit/moveit_tutorials #840: Cannot connect MoveIt with Franka Robot

Cody-Vu, 2025-01-29:
Hello everyone,

I use the Franka panda robot with ROS-noetic.
When I run "roslaunch franka_control franka_control.launch \ robot_ip:= \ load_gripper:=true \ robot:=panda“ and then “roslaunch panda_moveit_config move_group.launch” I get the error like in the picture. I just want to move the robot with ROS and furthermore with moveit in the future. Does anyone know how to get rid of that error so I can move the robot with MoveIt?

[image]

Thanks in advance!

rhaschke, 2025-01-29:
You need to start Franka's `effort_joint_trajectory_controller` either via the [command line](https://wiki.ros.org/ros_control/Tutorials/Loading%20and%20starting%20controllers%20through%20service%20calls) or via [rqt's controller_manager plugin](https://wiki.ros.org/rqt_controller_manager).

closed 2025-01-29 by rhaschke

Cody-Vu, 2025-01-30:
> You need to start Franka's `effort_joint_trajectory_controller` either via the [command line](https://wiki.ros.org/ros_control/Tutorials/Loading%20and%20starting%20controllers%20through%20service%20calls) or via [rqt's controller_manager plugin](https://wiki.ros.org/rqt_controller_manager).

So I loaded frankas effort_joint_trajectory controller via the command: 
rosservice call /controller_manager/load_controller "name: 'effort_joint_trajectory_controller'"
ok: True

But after I start it:
rosservice call /controller_manager/switch_controller "{start_controllers: ['effort_joint_trajectory_controller'], stop_controllers: [], strictness: 2}"
ok: True
I get the error in the terminal where I executed (roslaunch franka_control franka_control.launch \ robot_ip:= \ load_gripper:=true \ robot:=panda)
[image]

Cody-Vu, 2025-01-30:
> Stop posting images of your errors! Copy&paste those logs, which makes them searchable! Again, read the log:
> 
> ```
>   what(): libfranka: unable to set realtime scheduling: Operation not permitted
> ```
> 
> Googling this error [yields several relevant hits](https://www.google.com/search?q=libfranka%3A+unable+to+set+realtime+scheduling%3A+Operation+not+permitted&oq=libfranka%3A+unable+to+set+realtime+scheduling%3A+Operation+not+permitted&gs_lcrp=EgZjaHJvbWUyBggAEEUYOTIGCAEQRRg60gEHMzc1ajBqN6gCALACAA&sourceid=chrome&ie=UTF-8).

I apologize for the inconvenience. I haven't thought about it.

Cody-Vu, 2025-02-02:
@rhaschke I managed to load and start the effort_trajectory_controller. I tried to run move_group_python_interface_tutorial.py as a test. The first movement is executed successfull (the method "go_to_joint_state": https://github.com/moveit/moveit_tutorials/blob/master/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L175). But when it tries to execute "got_to_pose given" the eef pose: (https://github.com/moveit/moveit_tutorials/blob/master/doc/move_group_python_interface/scripts/move_group_python_interface_tutorial.py#L211) i get the following error and stops:

[ INFO] [1738327227.111203909]: Planning attempt 1 of at most 1
[ INFO] [1738327227.112855031]: Planner configuration 'panda_arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1738327227.113204200]: panda_arm/panda_arm: Starting planning with 1 states already in datastructure
[ INFO] [1738327227.136603127]: panda_arm/panda_arm: Created 5 states (2 start + 3 goal)
[ INFO] [1738327227.136653951]: Solution found in 0.023659 seconds
[ INFO] [1738327227.170385446]: SimpleSetup: Path simplification took 0.033677 seconds and changed from 4 to 2 states
[ WARN] [1738327231.822885230]: Controller 'effort_joint_trajectory_controller' failed with error GOAL_TOLERANCE_VIOLATED: panda_joint1 goal error -0.249576
[ WARN] [1738327231.823145022]: Controller handle effort_joint_trajectory_controller reports status ABORTED
[ INFO] [1738327231.823249959]: Completed trajectory execution with status ABORTED ...
[ INFO] [1738327231.829194384]: Received event 'stop'

Am I right that I have to adjust the tolerance in the <>controller.yaml? In the panda_moveit_config is only fake_controllers.yaml and simple_moveit_controllers.yaml. In franka_ros/config is default_controllers.yaml. Do you know which file actually gets used while running the moveit script and where I can adjust the tolerances?