moveit/moveit_tutorials #841: How to use planners in MoveItGroup python interface

Cody-Vu, 2025-01-29:
Hello everyone,

So I'm using MoveItGroup python Interface (ROS1-noetic) to plan paths for my robot. At the moment i can only plan with the default planner (with set_planner_id), which is RRT from OMPL. Do you know how to use other planners like STOMP, CHOMP or TrajOpt or even planning adapters? I also can't use the others which are in the ompl_planning.yaml (PRM, FMT,..). Or is it even possible? 
Also when I do the CHOMP tutorial the robot just drives through the obstacle without avoidance. And when I do the TrajOpt tutorial roslaunch panda_moveit_config demo.launch  pipeline:=trajopt it says it cannot find trajopt and no planner is initialized even tho I use the panda_moveit_config package where I changed nothing.

And do you know wether you can use impedance controller while using MoveIt?

Thanks in advance. Would appreciate it if someone can help me further!

rhaschke, 2025-01-29:
To switch the planning pipeline, use `MoveGroupInterface.set_planning_pipeline_id()`.
`MoveGroupInterface.set_planner_id()` selects a planner from a given pipeline. A list of available ids can be found in the rviz plugin: 
[image]

trajopt was never finalized. The package is currently [disabled](https://github.com/moveit/moveit/blob/master/moveit_planners/trajopt/CATKIN_IGNORE).

> And do you know wether you can use impedance controller while using MoveIt?

Sure, as long as it provides a trajectory controller.

Cody-Vu, 2025-01-29:
> To switch the planning pipeline, use `MoveGroupInterface.set_planning_pipeline_id()`. `MoveGroupInterface.set_planner_id()` selects a planner from a given pipeline. A list of available ids can be [...]

Thank you very much for the info. Please correct me if I'm wrong. So to clarify in pipeline_id I insert either OMPL/STOMP/CHOMP. This method then takes the configuration of the ompl/stomp/chomp_planning.yaml file which are in <robot>_moveit_config/config. And for OMPL I can furthermore set a specific planner (RRT, PRM,..). Does it also work for planning adapter for example CHOMP as post processor for STOMP or can I only use singular planner with the python interface?

rhaschke, 2025-01-29:
You summarized the planner selection almost correctly.

> Does it also work for planning adapter for example CHOMP as post processor for STOMP or can I only use singular planner with the python interface?

Actually, the pipeline_id names a pipeline, which needs to be loaded onto the parameter server. For example in panda_moveit_config, [multiple pipelines are loaded](https://github.com/moveit/panda_moveit_config/blob/da3252e54327d5b22aac6f1b1be2de75691e5def/launch/move_group.launch#L55-L82).
If you want to use ompl+chomp you need to load the [corresponding config](https://github.com/moveit/panda_moveit_config/blob/da3252e54327d5b22aac6f1b1be2de75691e5def/launch/ompl-chomp_planning_pipeline.launch.xml) as [documented in the tutorials](https://moveit.github.io/moveit_tutorials/doc/chomp_planner/chomp_planner_tutorial.html#using-chomp-as-a-post-processor-for-ompl).

Cody-Vu, 2025-01-30:
> You summarized the planner selection almost correctly.
>
>
> Actually, the pipeline_id names a pipeline, which needs to be loaded onto the parameter server. For example in panda_moveit_config, [multiple [...]

I guess I have some uncertainties and I'd be grateful if you clarify:

1. So you mean the method "MoveGroupInterface.set_planning_pipeline_id()" is not a method i can simply use in the python interface. To change the pipeline I have to first configure move_group.launch.

I think I misunderstood it. To break it down. I first have to adjust "move_group.launch" where all the planners are named are usable. For example in the following I tried to add stomp as a planner:

[image]

Here it includes (dirname)/planning_pipeline.launch.xml. This file loads the (dirname)/planning.yaml where the parameters are defined. In conclusion for a planner you have move_group.launch, (dirname)/planning_pipeline.launch.xml and (dirname)/planning.yaml.

2. Right now I simulate the robot with rviz using the command demo.launch. In this file I can set the pipeline as default.
But later I only need move.group.launch to move the real robot right?

3. In the stomp tutorial (https://moveit.github.io/moveit_tutorials/doc/stomp_planner/stomp_planner_tutorial.html) it says to install it from source but in the panda_moveit_config the xml and yaml file is already included. So I added the stomp planner in move_group.launch, set the default planner in the demo.launch to stomp but when i execute demo.launch it cannot find the stomp planner and no planner is initialized similar to trajopt. Is stomp also disabled? Also in move_group.launch there is no section "Planning Funcionality"

[image]

Thanks for your help so far!

rhaschke, 2025-01-30:
> Here it includes `$(dirname)/planning_pipeline.launch.xml`.  This file loads the `$(dirname)/planning.yaml` where the parameters are defined.

The latter file doesn't exist! Read the launch files!
- `move_group.launch` loads the parameters for [_several_ pipelines](https://github.com/moveit/panda_moveit_config/blob/da3252e54327d5b22aac6f1b1be2de75691e5def/launch/move_group.launch#L57-L81) onto the parameter server
- it does so, using the generic launch file `planning_pipeline.launch.xml` passing the `pipeline` name as an argument
- `planning_pipeline.launch.xml` in turn [loads `$(arg pipeline)_planning_pipeline.launch.xml` into the parameter namespace `$(arg pipeline)`](https://github.com/moveit/panda_moveit_config/blob/da3252e54327d5b22aac6f1b1be2de75691e5def/launch/planning_pipeline.launch.xml#L9)
- These files finally load various ROS parameters - specific for the given pipeline ([example for OMPL](https://github.com/moveit/panda_moveit_config/blob/da3252e54327d5b22aac6f1b1be2de75691e5def/launch/ompl_planning_pipeline.launch.xml#L17-L26))

> Right now I simulate the robot with rviz using the command demo.launch. In this file I can set the pipeline as default.
But later I only need move.group.launch to move the real robot right?

Yes. `demo.launch` [includes `move_group.launch`](https://github.com/moveit/panda_moveit_config/blob/da3252e54327d5b22aac6f1b1be2de75691e5def/launch/demo.launch#L56-L60). Later, you will launch your real robot and `move_group.launch`.

> Is stomp also disabled?

Yes.

You seem to be new to ROS. Why do you start with ROS1 instead of ROS2?

Cody-Vu, 2025-01-30:
> The latter file doesn't exist! Read the launch files!
>
> * `move_group.launch` loads the parameters for [_several_ [...]

Yes you are right. I'm completely new to ROS. For my thesis I'm using Franka and I need to use ROS because I want to connect it with my AI for object tracking. My supervisor told me to use ROS1, because that version was used before by his former students. But he himself hasn't operated with ROS thats why I'm kinda on my own. I also haven't managed to move the robot with ROS yet. That's why I'm open and grateful for any advice. So you recommend switching to ROS2?

rhaschke, 2025-01-30:
> So you recommend switching to ROS2?

ROS1 reaches EOL in May. Thus, I suggest switching to ROS2 on _new_ projects. If you are continuing a large existing project, you might want to stick to ROS1 (due to the migration effort).

Anyway, you should learn to read available docs. Issue trackers like this one are not for general questions. 
These should be better asked at https://robotics.stackexchange.com. Please, post real issues and bug reports here only.

rhaschke, 2025-02-03:
I cannot reproduce this error running latest branch of panda_moveit_config.

Cody-Vu, 2025-02-05:
Now I installed moveit new with panda_moveit_config. After installing `sudo` apt install `ros-noetic-moveit-chomp-optimizer-adapter` i run the command `roslaunch panda_moveit_config demo.launch pipeline:=ompl-chomp`. Then I received this error:

[move_group-5] process has died [pid 74843, exit code -6, cmd /home/catkin_ws/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/.ros/log/d4823688-e3e7-11ef-9e9b-3fafe23f3056/move_group-5.log].
log file: /home/.ros/log/d4823688-e3e7-11ef-9e9b-3fafe23f3056/move_group-5*.log
[ INFO] [1738777036.923888724]: Loading robot model 'panda'...
[ INFO] [1738777037.014892512]: Starting planning scene monitor
[ INFO] [1738777037.017188827]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1738777037.382558515]: waitForService: Service [/get_planning_scene] has not been advertised, waiting...
[ INFO] [1738777042.394963821]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[ INFO] [1738777042.395407643]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1738777072.424428386]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
[ INFO] [1738777072.425903768]: Constructing new MoveGroup connection for group 'panda_arm' in namespace ''
[ERROR] [1738777102.448950214]: Unable to connect to move_group action server 'move_group' within allotted time (30s)

[image]
I'd be really grateful when you can help me out.

Cody-Vu, 2025-02-06:
> Now I installed moveit new with panda_moveit_config. After installing `sudo` apt install `ros-noetic-moveit-chomp-optimizer-adapter` i run the command `roslaunch panda_moveit_config demo.launch pipeline:=ompl-chomp`. Then I received this error:
> 
> [ INFO] [1738777435.168877712]: Loading planning pipeline 'ompl-chomp' terminate called after throwing an instance of 'ros::InvalidNameException' what(): Character [-] at element [35] is not valid in Graph Resource Name [/move_group/planning_pipelines/ompl-chomp]. Valid characters are a-z, A-Z, 0-9, / and _. [move_group-5] process has died [pid 74843, exit code -6, cmd /home/catkin_ws/devel/lib/moveit_ros_move_group/move_group --debug __name:=move_group __log:=/home/.ros/log/d4823688-e3e7-11ef-9e9b-3fafe23f3056/move_group-5.log]. log file: /home/.ros/log/d4823688-e3e7-11ef-9e9b-3fafe23f3056/move_group-5*.log [ INFO] [1738777036.923888724]: Loading robot model 'panda'... [ INFO] [1738777037.014892512]: Starting planning scene monitor [ INFO] [1738777037.017188827]: Listening to '/move_group/monitored_planning_scene' [ INFO] [1738777037.382558515]: waitForService: Service [/get_planning_scene] has not been advertised, waiting... [ INFO] [1738777042.394963821]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()? [ INFO] [1738777042.395407643]: Constructing new MoveGroup connection for group 'panda_arm' in namespace '' [ERROR] [1738777072.424428386]: Unable to connect to move_group action server 'move_group' within allotted time (30s) [ INFO] [1738777072.425903768]: Constructing new MoveGroup connection for group 'panda_arm' in namespace '' [ERROR] [1738777102.448950214]: Unable to connect to move_group action server 'move_group' within allotted time (30s)
> 
> [image] I'd be really grateful when you can help me out.

If someone had the same issue. I had to rename `ompl-chomp_planning_pipeline.launch` to `ompl_chomp_planning_pipeline.launch` and execute it with `roslaunch panda_moveit_config demo.launch pipeline:=ompl_chomp`. Then It worked for me.

closed 2025-02-06 by Cody-Vu

rhaschke, 2025-02-06:
Thanks for reporting the error and the solution. I adapted the tutorials and panda_moveit_config appropriately.